bigtreetech/BIGTREETECH-TouchScreenFirmware #280: [FR] Printing screen with bar and fan

Hans007a, 2020-01-04:
Hello,

I like the 'old' printing screen, but I missed the fan value.
I change the printing.c to this:
[image]

Here is the source code :
`
#include "Printing.h"
#include "includes.h"

//1title, ITEM_PER_PAGE item(icon + label) 
MENUITEMS printingItems = {
//  title
LABEL_BACKGROUND,
// icon                       label
 {{ICON_PAUSE,                LABEL_PAUSE},
  {ICON_BACKGROUND,           LABEL_BACKGROUND},
  {ICON_BACKGROUND,           LABEL_BACKGROUND},
  {ICON_STOP,                 LABEL_STOP},
  {ICON_HEAT,                 LABEL_HEAT},
  {ICON_PERCENTAGE,           LABEL_PERCENTAGE},
  {ICON_BABYSTEP,             LABEL_BABYSTEP},
  {ICON_MORE,                 LABEL_MORE},}
};

const ITEM itemIsPause[2] = {
// icon                       label
  {ICON_PAUSE,                LABEL_PAUSE},
  {ICON_RESUME,               LABEL_RESUME},
};

#ifndef M27_WATCH_OTHER_SOURCES
#define M27_WATCH_OTHER_SOURCES    false
#endif

#ifndef M27_REFRESH
#define M27_REFRESH   3
#endif

static PRINTING infoPrinting;
static u32     update_time = M27_REFRESH * 100;

#ifdef ONBOARD_SD_SUPPORT
static bool    update_waiting = M27_WATCH_OTHER_SOURCES;
#else
static bool    update_waiting = false;
#endif

//
bool isPrinting(void)
{
  return infoPrinting.printing;
}

//
bool isPause(void)
{
  return infoPrinting.pause;
}

//
void setPrintingTime(u32 RTtime)
{
  if(RTtime%100 == 0)
  {
    if(isPrinting() && !isPause())
    {
      infoPrinting.time++;      
    }
[... 571 lines cut ...]
    }
  }

  if((infoPrinting.cur>=infoPrinting.size) && isPrinting())  // end of .gcode file
  {
    completePrinting();
  }
}

void loopCheckPrinting(void)
{
  static u32  nowTime=0;

  do
  {  /* WAIT FOR M27	*/
    if(update_waiting == true)                {nowTime=OS_GetTime();break;}
    if(OS_GetTime() < nowTime+update_time)       break;

    if(storeCmd("M27\n")==false)               break;

    nowTime=OS_GetTime();
    update_waiting=true;
  }while(0);
}
`

el-quique, 2020-04-22:
Hi. Which file do I edit to make it look like that?

oldman4U, 2020-10-20:
Hi.

Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 via ticket #732

Thank you

closed 2020-10-20 by Hans007a