bigtreetech/BIGTREETECH-TouchScreenFirmware #281: [BUG] (short description)

KookyKoalaJon, 2020-01-05:
### Description

When connection for Octopi is active on UART 3, thought the TFT 35 it connects for about 30 seconds to a minute then drops connection.
### Steps to reproduce
1. Wire RPI, to the RX and TX ports at the UART 3 connection on back of screen
2. With RPI Configured, for UART connect octpi it will connect
3. Watch 30's later Octopi Drops connections due to time out trying to talk to printer.

**Expected behavior**

When connected though external UART on TFT screen should stay connected the entire time

**Actual behavior**

Octopi drops connection every 30 seconds

Tried precompiled firmware it it connects, but drops out, when i try to build from source myself i can't seem to get the UART port working.

EsQueue81, 2020-01-06:
I can confirm this issue and ended up giving up. If you notice, it doesn't send the temperature requests. It will stay connected if you keep manually sending commands via octoprint terminal. I tried swapping the TX and RX pins and connected it to UART 4 and it does send and receive temperature but never connects. 

I never made a bug report as I am too pessimistic and you are a better person than I am.

KookyKoalaJon, 2020-01-06:
I couldn't ever get mine to connect via UART4 at all, either i may try and pin it into the uart1 port see if it makes any difference

EsQueue81, 2020-01-06:
> I couldn't ever get mine to connect via UART4 at all, either i may try and pin it into the uart1 port see if it makes any difference

That's the wifi connector right? I got nothing but see what your results are.

EsQueue81, 2020-01-07:
Here is a post from someone who just replied a few minutes ago. A thread where I thought that I got my pi connected successfully but failed to update. 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/231#issuecomment-571365256

Funny as I searched the entire folder for the word "UART" but it all seems to be right there in variants.h. I had to learn almost everything so I know that I didn't search it properly at the time.  I may try defining SERIAL_PORT_3 then SERIAL_PORT_4 as _USART1

Hotnozzle, 2020-02-04:
Hi,
I am struggling for a few days to get this setup to work: octopi on PI 4 with Ramps 1.4 with BigTreeTech TFT35 V3.0.25.  anybody knows how to go about? only one can work at a time but they cannot share the same port. Should I connect the PI to the TFT empty uart3? or should i connect the TFT to a serial port on the PI and get it to forward Gcode from TFT into the RAMPS card?

I am not sure if UART3 on TFT is already listening and forwarding on the BTT firmware....

any suggestion?

hfog, 2020-02-07:
I also have been entirely unable to get Octoprint to work when connected to the TFT through UART3 or UART4. I've verified that I can connect the pi directly to the printer over the GPIO pins but connecting those pins to UART3 (or 4) on the TFT does not appear to pass commands through. Looking through the code, it APPEARS that it is supposed to pass them through but I can see no evidence that it actually is.
Octoprint DOES work for me if I connect it directly to the printer over USB but I have to have the TFT in Marlin mode to avoid contention over the printer's serial port. I was really hoping for the pass through functionality to work and in a perfect world, I would expect this configuration to allow clean printing from Octoprint even in touch mode. (Even better would be if the touch mode showed Marlin status messages!)

hfog, 2020-02-12:
I have figured out how to keep Octoprint connected. If you go to the Serial Connection settings in Octoprint and set the "Temperature interval (autoreport)" setting to zero then Octoprint will use M105 to query temperature every few seconds. That keeps the connection alive.
Of course, this is not an optimal solution. What appears to be happening is that the TFT is not passing on auto-reported temperature ACKs to connected devices. Looking through the code, I'm not terribly surprised as it doesn't seem to really keep track of connected devices very well (if at all) and really only looks to pass ACK messages back to connected devices in direct response to commands.
There are still a number of issues connecting this way though. It appears that the TFT introduces a little bit of lag in the command processing and even occasionally garbles some of the commands. I will continue to look through the code and see if I can find solutions.
And you CANNOT use the emulation mode screen if you are connected over UART. Switching away from the touch screen puts you in a mode that doesn't contain the necessary polling loops. I believe that it should be possible to keep those loops running in emulation mode as well and I do plan to try that soon but this would still have the same limitations.

hfog, 2020-02-13:
OK. I've made a ton of progress. I have figured out how to allow Octoprint to connect through UART even in Marlin mode. What's more, I've fixed it so you can cleanly swap between Marlin and touch mode WHILE printing. The last thing I'd like to figure out is how to get the TFT to recognize when a print has been started on Octoprint so it can go into printing mode.
Once I've got that working I'm going to do a bunch of testing and then I'll push a PR so you all can play with it if you like.

EsQueue81, 2020-02-13:
> OK. I've made a ton of progress. I have figured out how to allow Octoprint to connect through UART even in Marlin mode. What's more, I've fixed it so you can cleanly swap between Marlin and touch [...]

Just letting you know that your efforts are appreciated.

hapklaar, 2020-02-15:
> OK. I've made a ton of progress. I have figured out how to allow Octoprint to connect through UART even in Marlin mode. What's more, I've fixed it so you can cleanly swap between Marlin and touch [...]

Will this also output all stats (XYZ position, temps, etc) on the TFT35 as would happen when printing from SD? That would be awesome. Hope you can also include a wiring schematic for Pi, SKR1.3 and TFT35 and settings for Marlin to get everything working.

hfog, 2020-02-16:
I've put my changes on https://github.com/hfog/BIGTREETECH-TouchScreenFirmware and I've created a PR here: https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware/pull/5.
Go ahead and play with it. It seems to solve most of the problems I was encountering. 
@hapklaar Yes, it correctly outputs position, temps, and fan. What it does NOT do is automatically detect when a print is happening from Octoprint over UART, but then neither does Marlin. It just knows it's being driven by a computer. It doesn't really know what the print state is. What this WILL do is correctly display M117 messages in the info window so if you're using the LCD message to display progress, that will show on the TFT info window.
Apologies for not forking from this repo but guruathwal's repo had a bunch of changes and fixes that made for a good jumping off point for this work.

hfog, 2020-02-16:
Note that you _might_ have to increase your timeout settings in Octoprint just a little bit. This is effectively introducing a middle-man in the equation since the TFT has to forward messages back and forth between Marlin and Octoprint.

hfog, 2020-02-16:
Oh. And it also forwards UART messages even when you're on the Marlin simulation screen. So you can leave Octoprint connected over UART even if you prefer to use the emulation screen.

hfog, 2020-02-16:
@hapklaar - I can't really include a wiring schematic because different TFT boards have different pin outs. The key is first to get Octoprint working when connected directly to your printer over serial 3 pin connector (tx, rx, gnd). Just pull the TFT/Serial cable and run the TFT using just the LCD cables in emulation mode until you get that working.  Google that and you should find a lot of tutorials out there. (I'll hunt one up for you in a minute.)
Once you've got that working, you should be able to reconnect the TFT/Serial cable and then wire Octoprint up to the UART3 (or UART4) port on the TFT, so long as it's running this firmware.  Remember that GND goes to GND, TX goes to RX and RX goes to TX.Won't matter if you're in emulation mode or touch mode. If you can't get Octoprint to connect, try swapping the pins around. They're low enough voltage that you won't damage anything if you get it wrong. I've found with the TFT pinout diagrams it isn't always obvious if they're listing the pins from the front or the back of the board. Probably my problem. :)

hfog, 2020-02-16:
Here's a post that helped me get Octopi wired up correctly: https://community.octoprint.org/t/simple-3-wire-serial-connection-from-pi3-to-skr-v1-3-board/10166

ssombra, 2020-02-18:
> @hapklaar - I can't really include a wiring schematic because different TFT boards have different pin outs. The key is first to get Octoprint working when connected directly to your printer over [...]

Hi I am a complete newbie regarding the use of repos and Github. How can I test your changes? do I have to clone your entire repo and use that instead of the files I used before (the repo from BTT) for compiling? If that's the case, how can I include the changes I made before I compiled?

hfog, 2020-02-18:
@ssombra Easiest thing to do would be to clone my repo into a clean folder and build it. Your changes won't be affected because they'll be in your own copy of the original repo in your other folder.
I wouldn't suggest merging your changes with mine because then you won't know for sure if anything you encounter is due to my changes or yours.

ssombra, 2020-02-18:
> @ssombra Easiest thing to do would be to clone my repo into a clean folder and build it. Your changes won't be affected because they'll be in your own copy of the original repo in your other folder. [...]

Thanks. Will try it and report if any issues.

ssombra, 2020-02-19:
SKR V1.3 with TFT35 V2.0. 
I flashed your firmware and followed the instructions to connect the rPi to the TFT UART3 but I can't make the connection. I get the following error:
"Error: No more baudrates to test, and no suitable baudrate found."
Any idea what is the problem?

hfog, 2020-02-19:
> SKR V1.3 with TFT35 V2.0.
> I flashed your firmware and followed the instructions to connect the rPi to the TFT UART3 but I can't make the connection. I get the following error:
> "Error: No more baudrates to test, and no suitable baudrate found."
> Any idea what is the problem?

If Octoprint isn't able to connect to the printer than you have a few things that it could be. In order for that to work, Octoprint has to be able to talk to the TFT (serial cable from that raspberry pi to the TFT), the TFT has to be listening and passing on messages (my firmware needs to be flashed) and the TFT has to be able to talk to the print (TFT - RS232 cable).

You've flashed the firmware to the TFT so that probably isn't it but just to be sure, check that the date and version number reported in the TFT's settings menu matches what you built from my code.

If the connection from the TFT to the printer is working then the TFT's touch menus should show that the printer is connected and you should be able to control the printer from the TFT. If you can home an axis then that's all working.

The most likely culprit is that the connection from the raspberry pi isn't working. Try hooking up the raspberry pi directly to the printer over the TFT port (you'll have to unplug the TFT's TFT/R232 cable). You'll need to connect the RX pin on the pi to the TX pin on your board, the TX pin on the pi to the RX pin on the board and a GND pin on the pi to the GND pin on your board. You'll also need to make sure that the UART port on the raspberry pi's pins are configured properly. Here's a page that describes the configuration changes on your raspberry pi: http://www.hobbytronics.co.uk/raspberry-pi-serial-port. With the TFT in emulation mode it will still be able to talk to your printer even with the TFT cable disconnected. If you can get Octoprint to connect directly to the printer over the serial port then you're halfway there. You should then be able to connect it to the TFT's UART3 port and the TFT to the printer and everything should flow through.

The last thing to check is to make sure that the TFT's baud rate and the printer's baud rate and Octoprint's baud rate all match.

Hope that all helps.

Scope666, 2020-02-19:
There's also pretty good info on pages 12-13 in this doc:

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/BIGTREETECH%20SKRV1.3%20guide%202019%20-%206.pdf

ssombra, 2020-02-20:
> > SKR V1.3 with TFT35 V2.0.
> > I flashed your firmware and followed the instructions to connect the rPi to the TFT UART3 but I can't make the connection. I get the following error:
> > "Error: No more baudrates to test, and no suitable baudrate found."
> > Any idea what is the problem?
> 
> If Octoprint isn't able to connect to the printer then you have a few things that it could be. In order for that to work, Octoprint has to be able to talk to the TFT (serial cable from the raspberry pi to the TFT), the TFT has to be listening and passing on messages (my firmware needs to be flashed) and the TFT has to be able to talk to the printer (TFT - RS232 cable).
> 
> You've flashed the firmware to the TFT so that probably isn't it but just to be sure, check that the date and version number reported in the TFT's settings menu matches what you built from my code.
> 
> If the connection from the TFT to the printer is working then the TFT's touch menus should show that the printer is connected and you should be able to control the printer from the TFT. If you can home an axis then that's all working.
> 
> The most likely culprit is that the connection from the raspberry pi isn't working. Try hooking up the raspberry pi directly to the printer over the TFT port (you'll have to unplug the TFT's TFT/R232 cable). You'll need to connect the RX pin on the pi to the TX pin on your board, the TX pin on the pi to the RX pin on the board and a GND pin on the pi to the GND pin on your board. You'll also need to make sure that the UART port on the raspberry pi's pins are configured properly. Here's a page that describes the configuration changes on your raspberry pi: http://www.hobbytronics.co.uk/raspberry-pi-serial-port. With the TFT in emulation mode it will still be able to talk to your printer even with the TFT cable disconnected. If you can get Octoprint to connect directly to the printer over the serial port then you're halfway there. You should then be able to connect it to the TFT's UART3 port and the TFT to the printer and everything should flow through.
> 
> The last thing to check is to make sure that the TFT's baud rate and the printer's baud rate and Octoprint's baud rate all match.
> 
> Hope that all helps.

I'll try this on the weekend. thanks.

hapklaar, 2020-02-20:
I will do the same if I have some time left.

hfog, 2020-02-20:
Great. My PR was merged into Guruathwal's repo last night so at this point you might want to pull his stuff instead of mine. https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware

ssombra, 2020-02-20:
> Great. My PR was merged into Guruathwal's repo last night so at this point you might want to pull his stuff instead of mine. https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware

Just to be sure, this works with the TFT35 v2.0, right?

hfog, 2020-02-20:
Yes.

ssombra, 2020-02-21:
> Yes.
> Should work.

It worked. I started with a new octopi image and followed the instructions. Now I can connect mi Pi to the TFT. Thanks

hapklaar, 2020-02-21:
Worked for me as well, this is sweet :)

One minor issue: I can only connect at 115200 or lower as the serial port on the Pi cannot be set to 250000. The nearest supported value is 230400, which is not supported by the TFT firmware.

On very complex prints it is possible to saturate a 115200 connection. Would it be possible to add 230400 to the list of speeds the TFT can work at?

hfog, 2020-02-21:
Glad to hear it!
But I'm a bit confused about your baud rate comment. I'm running my pi into the TFT at 250000 right now without any issue. You could pretty easily change the baud settings to support something else but 250000 should work just dandy. So long as Marlin's configured to work at that speed all should work end to end.

hapklaar, 2020-02-21:
Ah my bad. I thought this was because I wasn't able to set the serial speed on my Pi to 250000 with "sudo stty -F /dev/ttyAMA0 250000" but that was a classic red herring then :)

Just checked and speed in Marlin is indeed still set to 115200. Will recompile with 250000 and try again.

ssombra, 2020-02-24:
I've been testing this configuration and in terms of communications it works good; however, I have noticed that when printing through Octopi the printer makes some small pauses, particularly noticeable during circles, which creates some small blobs. Have you noticed it?

hfog, 2020-02-24:
Yes. The TFT definitely introduces a little bit of lag which can adversely affect prints with lots of short rapid movements. You definitely should enable RAPID_SERIAL_COMM in config.h but even that won't eliminate the problem entirely. You can alleviate some of it by printing with the TFT in emulation mode but you might still encounter some problems on particularly fast and busy prints. There will need to be some additional work done to optimize the communication path to completely eliminate this. So far I've not been successful in getting rid of it entirely.

ssombra, 2020-02-24:
I'll try with RAPID_SERIAL_COMM; unfortunately my TFT doesn't have an emulation mode (V2.0).

ssombra, 2020-02-24:
> I'll try with RAPID_SERIAL_COMM; unfortunately my TFT doesn't have an emulation mode (V2.0).

Isn't it already activated in your repo?

hfog, 2020-02-24:
> > I'll try with RAPID_SERIAL_COMM; unfortunately my TFT doesn't have an emulation mode (V2.0).
> 
> Isn't it already activated in your repo?

In my code it is but I don't remember what Gurmeet did with it when it was merged and who knows what others might have done with it after they pulled the code.

ssombra, 2020-02-24:
Ok, I asked because I am still using your code :( so no hope to improve it.

hfog, 2020-02-24:
Not yet. I think there's room to improve but I certainly haven't had time to do it yet. Wish I had a better story to tell you. We're all in the same boat here. Clearly the firmware on this thing has a long way to go before it unlocks its full potential.
If you've got a particularly troublesome print you can direct connect to the printer and unplug the TFT/RS232 cable. That will eliminate any variables introduced by the TFT in the system. You'll have to run the TFT in emulation mode. It's not a great solution but until the firmware can be fully optimized it's the best option. I've kept my USB cable plugged in to the printer for that very reason. I haven't given up hope on the TFT yet but for particularly troublesome prints, it's nice to have a backup option.

Scope666, 2020-02-24:
You know I've been thinking about this issue while I wait for my replacement TFT to arrive from China.  I'm thinking it might make sense to mount the TFT in it's own "holder" and keep it to the left, in a case where it's easy to unplug the serial cable.  It's not a sexy solution but it would solve the problem while printing from Octoprint.  (keeping the stock screen in the factory holder for "Marlin Mode")

ssombra, 2020-02-25:
> Not yet. I think there's room to improve but I certainly haven't had time to do it yet. Wish I had a better story to tell you. We're all in the same boat here. Clearly the firmware on this thing has [...]

Yes of course, I understand. Thanks for all the time you put on this and all the support for us newbies. Until further improvements, I think I'll print directly from Octoprint via USB.

Zuru1, 2020-04-14:
This sounds interesting but wanted to make sure I understod it correctetly. 
With this setup, you end up using the TFT35 V2 or V3 as an octoprint display screen instead of the usual printer screen right?

electricJP, 2020-04-14:
No. Connecting the PI to the TFT allows you to have the TFT actually display printing information. If PI is connected via USB, then TFT will only display coordinates and temperature.

Zuru1, 2020-04-14:
> No. Connecting the PI to the TFT allows you to have the TFT actually display printing information. If PI is connected via USB, then TFT will only display coordinates and temperature.

A few follow up questions, 
- When connected this way, does TFT still need to be connected to the printer board?
- Would you be kind to post pic of what is displayed when its connected this way? 

Thx u.

electricJP, 2020-04-14:
Yes. You have to connect the tft to the printer board whatever solution you choose to connect the PI.
[image]

Zuru1, 2020-04-14:
@electricJP 
Sorry forgot to ask one more thing, this is only aimed for marling firmware and not klipper right?

electricJP, 2020-04-14:
> @electricJP
> Sorry forgot to ask one more thing, this is only aimed for marling firmware and not klipper right?

I do not have any practical knowledge of Klipper, sorry I can't answer you.

paul-1, 2020-05-09:
@hfog Can you spot anything specific regarding uart connection between your master branch and what ended up getting merged here.

I can connect to uart3 with firmware built from your branch, but not with the current master here.  So much has changed.

Ahmedk1995, 2020-06-13:
@hfog sorry to bug you but I had a question.
I was able to get octoprint to connect to the board through the tft using the 3 wire connection method RPi>TFT>Board
However i can only use touch mode with octoprint, as soon as I switch over to marlin mode the print stops until I move back to touch mode, once I move back to touch mode the print starts up again and it gives me a warning saying "line number is not last line number+1"
Do you have any idea what's going on here, I thought we should be able to cleanly switch between touch and marlin mode now and it seems like I can't.

hfog, 2020-06-13:
Looks like maybe you haven't enabled CLEAN_MODE_SWITCHING_SUPPORT.
Note that it's been quite a while since I've pulled the latest code so I'm not sure what the current state is.

Ahmedk1995, 2020-06-14:
> Looks like maybe you haven't enabled CLEAN_MODE_SWITCHING_SUPPORT.
> Note that it's been quite a while since I've pulled the latest code so I'm not sure what the current state is.

That was exactly it, had to learn how to compile my own copy of the firmware since the preloaded .bin didn't have it enabled.
Thank you for your quick reply and great work, this feature works quite well!

terghalin, 2020-06-15:
I think that it's better to print from SD/USB card from TFT35 then or just forget about touchscreen mode and print via Octoprint. Too bad it's still that way.

Ahmedk1995, 2020-06-15:
> I think that it's better to print from SD/USB card from TFT35 then or just forget about touchscreen mode and print via Octoprint. Too bad it's still that way.

It's actually been working great ever since I enabled that option above. Been printing through octoprint with both touch mode and marlin mode fully functional

terghalin, 2020-06-16:
Well, even if you enable thosw 2 options that were mentioned (SERIAL_ALWAYS_ON and RAPID_SERIAL_COMM), you will still experience small pauses during fast prints. Because even a small pause can cause severe issues for model you are printing, I'm going to pass with OctoPrint -> TFT- > Marlin setup for that reason.

hapklaar, 2020-06-16:
> Well, even if you enable those 2 options that were mentioned (SERIAL_ALWAYS_ON and RAPID_SERIAL_COMM), you will still experience small pauses during fast prints. Because even a small pause can cause severe issues for model you are printing, I'm going to pass with OctoPrint -> TFT- > Marlin setup for that reason.

Unfortunately that is also my experience. It's a nice gimmick and would be great to have as a reliable default way for printing, but on my end it often doesn't work, prints are not as smooth and if they print they print slower than with a direct serial connection from OctoPrint to SKR. It looks like the TFT35 is not able to handle the serial stream very well and introduces tiny pauses in the comms flow.

terghalin, 2020-06-16:
> Unfortunately that is also my experience. It's a nice gimmick and would be great to have as a reliable default way for printing, but on my end it often doesn't work, prints are not as smooth and if [...]

BTW, what kind of Raspberry you have? Connection through miniUART can cause problems. I've just changed RPi 3 B+ kernel settings to point on UART PL011 (from bluetooth) instead of miniUART, going to test it soon.

kind3r, 2020-06-16:
Any specific model to test with ?
I just finished installing a TFT 35 on my 32bit modded Artillery Genius. I'm runing OctoPrint on a PI 4B with 4G RAM and doing a Marvin and not noticing anything special so far. The PI is connected via uart 5 (GPIO 12&13) to the uart 3 of the TFT.

terghalin, 2020-06-16:
AFAIK, RaspberryPi from 1 to 3 has miniUART on GPIO 8 and 10 by default, which will certainly cause problems. On RPi 4 there is 6 UART's and only one is mini.

kind3r, 2020-06-16:
If all else fails I think a USB FTDI module would also work just fine if you can't spare the one UART you have on the PI.

kind3r, 2020-06-17:
So I have my PI 4 running Octoprint connected to the TFT 35 which is connected to the MKS SGEN mainboard via serial and EXT3 (the Creality one so I don't use both my EXT ports on the MB). All running smoothly for now except for the `M117` messages not showing on the TFT status screen, there seems to be a message stuck there (well not stuck, it is scrolling) from boot: 
```
(i) echo: 
Fade Height 3.00
```
I checked and `M117` messages are being send and they are displayed in Marlin mode.

I also looked through the code and I don't see anything wrong as far as I can tell (comands received by the TFT from UART are put in the queue and ran just like any other command so the M117 should update the status screen message).

Any ideeas ?
PS: using @guruathwal 's fork build by myself with no changes.

kind3r, 2020-06-17:
Ok, I have identified one issue in `interfaceCmd.c:530`.
```C
   strncpy(message, &infoCmd.queue[infoCmd.index_r].gcode[cmd_index + 4], CMD_MAX_CHAR);
```
since `cmd_seen` was not used, `cmd_index` has the value of the previous result, truncating the M117 message.

So the line shoud be:
```C
   strncpy(message, &infoCmd.queue[infoCmd.index_r].gcode[4], CMD_MAX_CHAR);
```
just skip the first 4 bytes of `M117`.

This brings me to the next issue. Since Octoprint issues commands with line number and checksum, those commands are stored as such in the queue and therefor the parser in `sendQueueCmd()` will efectively ignore everything. That is the reason why M117 are totally ignored.

kind3r, 2020-06-24:
I have made some changes to the parsing of stored commands to fix the issues with `M117` and also fix the lines not being parsed when printing to Octoprint because of the existence of line numbers and thus the display not being updated with proper information. I put my changes in my repo [here](https://github.com/kind3r/BIGTREETECH-TouchScreenFirmware) and also made a PR against [guruathwal's branch](https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware).

[18 more replies]

closed 2021-03-03 by stale[bot]