bigtreetech/BIGTREETECH-TouchScreenFirmware #282: [BUG] Very poor print quality since version 3.0.25.1

gneu42, 2020-01-06:
### Description

Since version V3.025.1 of TFT35 V3.0, I notice a big degradation of the print quality. 
So I ran some test print with a few software version as well as a print in Marlin Mode.
I noticed that for version V3.0.25.1, during all round perimeters, the print head was stopping twice for a very short time in each perimeter. That's what cereates the blobs around the part.
For Test 1 and 5 (V3.0.25 and marlin mode) all perimeters are printed smoothly in one shot with no stops.

Here are the results of my test prints

Test conditions :
Printer : Creality Ender 3 Pro
Mainboard : SKR 1.3
Screen : TFT335 V3.0
All tests Printed on the same day one after the other.
All tests printed from the same gcode file.
All tests printed with same filament.
 
1. V3.0.25 
   Downloaded from : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware	
   download date : DEC 25 2019
   Source files date : Dec 25 19 5:10
   Photo : 1.jpg
   Quality : Very good

2. V3.0.25.1 
   Downloaded from : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware	
   download date : DEC 30 2019
   Source files date : Dec 29 19 17:07
   Photo : 2.jpg
   Quality : Bad
 
3. V3.0.25.1 
   Downloaded from : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware	
   download date : Jan 3 2020
   Source files date : Jan 2 2020 02:17
   Photo : 3.jpg
   Quality : Bad

4. V3.0.25.1 
   Downloaded from : https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware	
   download date : Jan 4 2020
   Source files date : Jan 3 2020 08:22
   Photo : 4.jpg
   Quality : Bad

5. V3.0.25.1 Marlin Mode
   Downloaded from : https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware	
   download date : Jan 4 2020
   Source files date : Jan 3 2020 08:22
   Photo : 5.jpg
   Quality : Very good

*************************************
**Expected behavior**
Good quality print

**Actual behavior**
Bad print quality

### Additional Information
Photo 1
[image]

Photo 2
[image]

Photo 3
[image]

Photo 4
[image]

Photo 5
[image]

All prints from 1 to 5
[image]

guruathwal, 2020-01-09:
Did you noticed any stutter while printing?  did the printer loked like it stopped for few moments at points where oozing is visible?
btw your bad prints are way better than mine. which slicer do you use?

gneu42, 2020-01-09:
Yes, I saw the printer stopping twice for very short time during perimeter.
Those prints were sliced with PrusaSlicer 2.2.0-alpha2. It seems to make better prints than Cura. But does not have all the bells and whistle that Cura has.

arut16, 2020-01-14:
I have noticed the same issue with touch mode . Will try a new print in Marlin mode.
My config:
Ender 3 Pro
BTT SKR 1.3
BTT TFT35 V3.0.25
2 MOSFETS (1 HOTEND+1 HOTBED)
BTT Relay 1.2

JWolfJr, 2020-01-20:
I'm getting the same. 
My setup:
Ender 3
BTT SKR mini E3 1.2
BTT TFT35 V3.0 with latest firmware
Raspberry pi 3B+ octoprint
As I was getting the same results.  I ran a print from the touch screen with a file on a usb stick to bypass running off my octopi.  It looked fine.  I believe my issue is when printing from the octopi while on the touch screen mode I get the exact same blobs/zits as seen in the pictures.  And you can actually see the hot end stutter/hesitate.  As soon as I switch to the simulated screen, even mid-print, the problem goes away.  My thoughts were the tft port and the usb port on the SKR mini E3 are some how connected????

arut16, 2020-01-20:
I think I found what was the problem for me : my 2 TFT cables from skr was too long (I changed them to longer ones). I replaced them to the stock ones and no more printing issue described here (I moved the TFT screen from right to left at the same level as SKR board).

gneu42, 2020-01-21:
After reading all your comments, I decided to give a try to octoprint.
I ran the same test print using octoprint with the TFT35 in touch mode (vith version V3.0.25 that gave good quality during my first tests) and the result came out with lots of blobs like the other bad prints.
Than I print again with octoprint but with the TFT35 in marlin mode, and the print came out with very good quality.
I think the serial communication makes the problem. It seems that the buffer gets empty when doing the very short moves in circles. I even tried to increase the buffer size in Marlin, but never geet good quality in touch mode.

Conclusion : I think I will remove the TFT 35 and put the original creality screen, since to get good quality print the TFT 35 has to be in marlin mode !!! Unless BigtreeTech comes up with a fix in a futur software release.

guruathwal, 2020-01-26:
@gneu42 just push the PR for the issue. can you try to check if it fixed the issue? you will need to uncomment `#define RAPID_SERIAL_COMM` in configuration.h

you can get the source from here till the PR is merged: https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware

gneu42, 2020-01-27:
Hi,

Well, I spent quiet a lot of time testing this new version with RAPID_SERIAL_COMM enabled. The test print came out very good. Then I desabled RAPID_SERIAL_COMM, and the test print came out also very good. I did the test with menu_list_mode enable and disable. All prints are good !!!
I could not understood what happened, then I loaded back the version that was making bad print on january 6 together with the marlin version of that date. And I could not have a bad print. I do not understand why everything look ok now and did not work ok on january 6.
Since i have a raspberry PI installed with octoprint, I did a test print using octoprint. And the result came out bad even with RAPID_SERIAL_COMM enabled. I suppose that the problem is related to Marlin that have 2 serial ports to manage. Although with a 32 bits processor on the SKR1.3 everything should be smooth.
I did not see any problem with navigating into other display while printing when RAPID_SERIAL_COMM is enabled. I htink you can leave this mod in the softare.

I really do like the new print display. Great job, but I noticed some other small problems related to model preview.
In list mode and non list mode, when printing, the model preview does not appear in the lower left corner like it is supposed to, but in the second position on the top row, like it was in the version I test on january 6. The preview is somtime the wrong preview (See attached video and picture)!

In non list mode, the model preview on the SD Card are only displaied correctly the first time the card is accessed, after display reboot. When I navigating in the SD card, and come back one step, (almost) all the model previews are gone (see attached video). The second preview on the top row seems to be very often a black square (Background icon).

[Wrong model preview.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4117625/Wrong.model.preview.zip)
[SD Card Navigation.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4117628/SD.Card.Navigation.zip)

[image]

guruathwal, 2020-01-27:
@gneu42 this is the best explanation I have read till now..👍🏼.. good to know that problem has vanished..
The preview image location problem has been fixed.. you must have downloaded the source before the fix.. re-download the source should fix it.. I will try looking in the preview image in icon mode problem..

guruathwal, 2020-01-28:
@gneu42 model preview fixed here : https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/302/commits/ce4c8342a6441e69b2a4e85ab87783d55824fe1a 
can you chek if it is solved?

gneu42, 2020-01-29:
Hi,
Thanks for the fix, so far everything looks great. Navigation through files is a dream now. Preview are displayed very well now.
I still have to try subdirectories and print a few things. I'll keep you informed tonight or tomorrow.

Now i have to make all the preview files. By the way i have problem using the bigtree-tool (model builder). I can get it to work in english or french, All i have seems chinese language. I can't open the PDB files, and all I can do with are the basic task : open a stl and save it as bmp. I see a lot of exe files and language the sub-directories of the distributed zip file, but have no idea of what to do with them.
Any idea where I can find a good user manual for that utility ?

gneu42, 2020-01-30:
Hi,
Yesterday and this morning i did some checks and prints. All was ok even with files living in 2 levels of sub-directories.
Preview were always correctly displayed on SD card files display and also on printing menu.

I thank you for your very good work fixing those issues.

Best regards from Belgium

arut16, 2020-01-30:
> Hi,
> Yesterday and this morning i did some checks and prints. All was ok even with files living in 2 levels of sub-directories.
> Preview were always correctly displayed on SD card files display and [...]

Hi everyone,

I applyed the same fix but previews of gcode files in 2nd level of sub-directories are not show up... If i put gcode files with bmp files on root of the SD-card then previews are working... don't understand what I'm missing here...

Any ideas ?

My config : SKR 1.3+TFT35 v3

closed 2020-01-31 by gneu42

Scope666, 2020-02-01:
I know this was closed already, but like gneu42 , I have BAD blobbing if I try to print from Octoprint, this is a new problem that didn't exist before I added the TFT35 V3 E3.  It doesn't matter if I'm in Marlin mode or Touch Mode.  I can also visually see the stuttering.

guruathwal, 2020-02-01:
> Maybe if we had a way to go into settings on the TFT and tell it to kill comms or something?
@Scope666 there is an option in TFT setting to disconnect serial.

Scope666, 2020-02-01:
> > Maybe if we had a way to go into settings on the TFT and tell it to kill comms or something?
> > @Scope666 there is an option in TFT setting to disconnect serial.

Do you think this would solve the problem, or is it just a Marlin issue managing 2 serial ports?  Thanks again for all your help today!    :)

Scope666, 2020-02-02:
I think I might have solved this issue, at least this trick has been working for me when printing from Octoprint:

1.  You have to be on the screen that appears after you click disconnect. "Now you can control the printer by your computer!"   I believe if you touch the screen again it re-connects.

2.  You can't go into Marlin mode at all.  Once you do, even if you go back to TFT mode, I believe it is staying connected behind the scenes.  I've noticed that once you've switched to Marlin mode once, after that it switches instantly, that makes me think it's staying connected.  (1st connect takes a lot longer)

JWolfJr, 2020-02-02:
I would like to try this, where does it say disconnect? On the touch screen? Then try printing from Octoprint?

Scope666, 2020-02-02:
> I would like to try this, where does it say disconnect? On the touch screen? Then try printing from Octoprint?

Yeah, on touchscreen, go to MENU --> SETTINGS --> DISCONNECT right after freshly powering on the printer.  Once on the screen that says "Now you can control... "  don't touch the screen again or it reconnects.  Try printing from Octoprint like this and report back.    :)

JWolfJr, 2020-02-02:
Definitely will do! I'm in the middle of a 15hr print right now.  Nintendo Switch case for my son. Can't wait to try, been looking for a work around.  Thanks for getting back.

JWolfJr, 2020-02-04:
Disconnecting works, too bad you can't go back to the home screen.  Printed small circles with out problem.  Thank you for posting, always saw the disconnect icon, never pressed it.

Scope666, 2020-02-04:
The next thing we have to test is starting up in Marlin mode clean (not going to TFT at all) and see if we can still print ok from OctoPrint.  The idea is to see if the serial connection between the MB and TFT causes any issues even if we're not in TFT mode.

JWolfJr, 2020-02-04:
I have been printing all weekend from marlin as the start up screen with no issues.  For this test was the first time I put it back onto the tft mode for over a week now.

Scope666, 2020-02-04:
> I have been printing all weekend from marlin as the start up screen with no issues. For this test was the first time I put it back onto the tft mode for over a week now.

So the print quality issue is only printing from Octoprint with TFT in Touch mode and active?  Just making sure I know what to avoid.

JWolfJr, 2020-02-04:
Yes! That is correct. While testing just earlier, about a quarter through the print I pressed back twice to get to the main screen and the steppers started stuttering and leaving blobs/zits.  I immediately went back and disconnected, the problem cleared up, no more stuttering!

Scope666, 2020-02-04:
Very interesting ... hopefully they can make some improvements on the Marlin side to prevent this in the future.  Phase 2 would be to show print status via gcode M73 (OctoPrint plugin) like you can now in Marlin mode.

JWolfJr, 2020-02-04:
That would be nice, I do like the top bar status of the bed and hotend temps.  Its a useful and nice touch.  I do have the display layer progress plugin for octoprint, now that you said something I just realized it doesn't show in tft mode.  Guess I'll keep checking on the progress of that. Thanks for bringing the disconnect feature to my attention.  I suppose you can do what you need to in the first layer, like baby stepping for example, then just disconnect and finish the print.  Have to go to work now.  Thank you again.

Scope666, 2020-02-04:
> That would be nice, I do like the top bar status of the bed and hotend temps. Its a useful and nice touch. I do have the display layer progress plugin for octoprint, now that you said something I [...]

Glad I was able to help.  Thanks for tipping me off to the pure Marlin mode thing, glad to know I can still use that with Octoprint.  For not I think I'll use TFT for messing  with the machine, then switch to Marlin mode when I'm about to start a print.  I have the double tap button shortcut to probe offset which is pretty convenient.

guruathwal, 2020-02-04:
@JWolfJr 
> That would be nice, I do like the top bar status of the bed and hotend temps. Its a useful and nice touch. I do have the display layer progress plugin for octoprint, now that you said something I just realized it doesn't show in tft mode. Guess I'll keep checking on the progress of that.

You can enable the top status bar by going to `Settings > Feature Settings > Persistent Status Info`

@Scope666 
> @guruathwal Do you have any ideas on how I can print from Octoprint and not have the TFT ruin the print?

Did you tried connecting all of them in series? like `MainBoard > TFTcontroller > OctoPi` ?

Scope666, 2020-02-04:
> 
> Did you tried connecting all of them in series? like `MainBoard > TFTcontroller > OctoPi` ?

No, I'm not sure how to do that.  Right now it's a standard USB cable (with power pin covered) between MB and Pi.

guruathwal, 2020-02-04:
@Scope666 you will have to remove the USB serial and connect the RPi's Hardware serial ports to the serial pins on the ESP ports of the TFT. (you may need to enable hardware serial in RPi config)

it will be just like Hulk version of the ESP01 🤣

gneu42, 2020-02-04:
@guruathwal This sound interresting. Is there a procedure somewhere to get it done ?

robbycandra, 2020-02-04:
I believe it is because the marlin is too busy handling two serial port. Both send gcode to marlin which makes marlin have to work for UsbSerial and MSerial. For example both Octoprint and TFT send "M105". This command is asking Temperature Update. This simple command From Octoprint will distrub the printing gcode stream from octoprint. 

The solution maybe is to stop send M105 to Marlin.

Scope666, 2020-02-05:
I think part of it is you have to fresh boot into Marlin mode.  If you just switch to Marlin mode after being in touch mode it acts like the communications are still going on under the hood.

guruathwal, 2020-02-05:
@gneu42 just search on google.. you will fine plenty of tutorials.. like this one.. https://www.raspberrypi.org/forums/viewtopic.php?t=213133

Scope666, 2020-02-05:
@guruathwal   So I just had a print running from Octoprint fail 6 hours in, I had booted the machine in Marlin mode and never entered touch mode.  Can you confirm that there is no communication over the RS232 / Serial / TFT cable between the TFT and Motherboard in Marlin mode?  If there's any chance that it IS communicating that should be disabled in Marlin mode, or Octoprint will never work.

The last line in the Octoprint log when it failed:
`SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)`

They specifically say here that the Pi should be the only thing sending serial when printing:

https://community.octoprint.org/t/octoprint-randomly-loses-connection-to-the-printer-with-a-serialexception/228

guruathwal, 2020-02-05:
@Scope666 regarding the serial comm. When switching to marlin mode the firmware clearly switches off the serial comm. By calling  `Serial_ReSourceDeInit()` and  `Serial_DeInit()` to turn off serial comm.
This should have same effect as it would when click on disconnect button.

Scope666, 2020-02-05:
Thanks for the quick reply.  I'm stumped then, I never had this happen before I connected the TFT35 E3.  Can you think of anything else that might cause this?  The Marlin emulation itself perhaps?

If I can't solve this I'll have to return the TFT, which would suck, because I love touch mode for things like moving axes, homing, bed leveling, etc.  It's a great UI.

guruathwal, 2020-02-05:
I don't think its a software issue.
It might be a faulty usb cable. If you are using the blue usb cable which came with the board then you might need to change it as the have very weak wires. I received two boards with faulty usb cable.
Also you should atleast try connecting the RasPi through the TFT rather than connecting directly to the board.

Scope666, 2020-02-05:
I'm willing to give this a shot.  From the manual it looks like TX goes to RX and vice-versa plus a ground connection.  Does anything need to be changed in either Marlin or TFT firmware to make this work?  I'm trying to understand how the gcode goes from the Pi to the TFT and then the Motherboard.

[image]

guruathwal, 2020-02-05:
Tx -> Rx  ,  Rx ->Tx  Gnd -> Gnd

TFT automatically passes any gcode received from printer to the serial port for ESP and vice versa, you do not need to change any thing in marlin or TFT.

Scope666, 2020-02-05:
I found a proper guide if it helps anyone else.  It's for the SKR 1.3 but I would imagine the Pi changes would apply for other boards:   https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/BIGTREETECH%20SKRV1.3%20guide%202019%20-%206.pdf

Pages 12-13

JWolfJr, 2020-02-05:
I'm sorry, I'm a little slow.  Is this supposed to be a work around for the stuttering when printing with the tft screen is enabled?  Wont this disable touch screen functionality?

guruathwal, 2020-02-06:
@JWolfJr No.. we are not trying to disable the TFT.. we are trying to connect the Raspi to TFT instead of the board, so we can have multiple devices connected to single serial port on board.

JWolfJr, 2020-02-06:
Ok, just so I'm following correctly.  We are connecting the tft cables to the raspi now instead of the tct display?! So the tft display will only be connected to the MB through AUX3?

guruathwal, 2020-02-06:
@Scope666 you can enable Dry run with `M111 S8` to test long run gcode files without extruding.

@JWolfJr 
This might help you to understand:
[image]

JWolfJr, 2020-02-06:
Ok, sorry.  I failed to read a few posts back.  RPi tx/rx/gnd to the wifi/esp ports of the TFT? Regular RS232 connection to the MB?  Have you tried this yet? Cleaned up the prints?

JWolfJr, 2020-02-21:
Sorry for the delay, finally got around to testing. I am still getting the blobs/zits with the stepper stutter in tft mode.  Seems worse actually, doing it straight lines now!  Stops soon as I switch to the emulated mode.  I did download and compile guruathwal's version.  Also a few icons are missing, I know it's not part of this thread.

thinkyhead, 2020-02-26:
Can anyone sniff the serial traffic between the TFT and the firmware to see if it's spamming?

Scope666, 2020-03-03:
I see they just pushed this commit for Marlin, wondering if it will improve things:

https://github.com/MarlinFirmware/Marlin/pull/17032

gonzaloalbito, 2020-04-01:
> I see they just pushed this commit for Marlin, wondering if it will improve things:
> 
> [MarlinFirmware/Marlin#17032](https://github.com/MarlinFirmware/Marlin/pull/17032)

I'd been running Marlin 2.0.5 (which is supposed to include the mentioned fix) from two weeks ago and the problem is still here.
I'm able to reproduce the problem no matter octoprint is connected or it's not.
On the other hand touch screen is out of sync with marlin after a few minutes of machine start up. It's able to read some values but unable to perform any operation like fan control, movement, change temterature... I don't know if it's related but i mention it just in case...