bigtreetech/BIGTREETECH-TouchScreenFirmware #283: Error after cancel printing

andrzej1974, 2020-01-07:
Hello
In new compilation TFT35 V3.0.25.1 after push stop printing I get this message 
[image]

guruathwal, 2020-01-08:
which was the last commit where it was working without error?

andrzej1974, 2020-01-08:
I'm not sure but I have files from beginning January end working good.

andrzej1974, 2020-01-08:
this was after
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/266

śr., 8 sty 2020, 07:25 użytkownik Andrzej <andacz6@gmail.com> napisał:

guruathwal, 2020-01-08:
> this was after #266

did you used the source before this commit or after it?

andrzej1974, 2020-01-08:
after this and was good

guruathwal, 2020-01-08:
are you sure? #266 was never merged.

andrzej1974, 2020-01-08:
this was after remove blue progres percentage bar and apply long filename onboard files. I think is after last changes. But I don't remember exactly.

guruathwal, 2020-01-08:
can you send/upload the last source code you have which does not gave this error?

andrzej1974, 2020-01-08:
no problem, but today afternoon

andrzej1974, 2020-01-08:
this is a last good compilation what i have.
https://drive.google.com/file/d/1orb_UP5fUdR-TcEd7Em4Bq4ENQ-UAABy/view?usp=sharing

guruathwal, 2020-01-09:
can you share the gcode file you tried to print?

andrzej1974, 2020-01-09:
https://drive.google.com/open?id=1pAU6V5WBt1x4qPxn1hJdLkohNP23UA3C

there is few, but i try all files.

oldman4U, 2020-09-03:
Hi.

What is the status of the reported issue?

@andrzej1974 Have you been able to fix the issue in the meantime?

Please let us know

closed 2020-12-15 by stale[bot]