bigtreetech/BIGTREETECH-TouchScreenFirmware #286: White screen with garbage after uploading firmware

jimhiggs, 2020-01-08:
I received tft3.5 v2.0 today and plugged it into my SKR Pro v1.1 worked a treat. I perhaps wrongly assumed the software on it was just a demo so I followed the instructions to update the firmware.
It ticked through the percentage complete then white screen with just a mess. Touch screen seems to do something. I then downloaded  BIQU_TFT35_APP1_V2.0.25.bin from here and put that on the sd card and reset.... Same thing happened. So is it junk now?

jimhiggs, 2020-01-08:
[image]

Hukuma1, 2020-01-08:
You need to copy the TFT35 folder to your SD card as well, not just the .bin firmware. The folder contains the fonts and images (your font is clearly the issue in the photo).

jimhiggs, 2020-01-08:
Thanks Hukuman1. You are my saviour. I did that and I now have functioning icons but some of them eg Heat/Fan are empty ie no picture?

Hukuma1, 2020-01-08:
Can you post an updated photo?

jimhiggs, 2020-01-08:
[image]

Hukuma1, 2020-01-08:
Check your TFT35\bmp folder. Do you have Heat.bmp and Move.bmp? If so, try to do a fresh copy of it to your SD card and overwrite the bmp files. (Simply rename TFT35CUR back to TFT35)

jimhiggs, 2020-01-08:
Ok did that and yes the files were on the the sdcard and I viewed them ok on pc and when it was updating the icons i watched an none were missing. Just missing in the display. There are others missing too.
[image]

Hukuma1, 2020-01-08:
Oh you are using unified material menu. Those require the other icon set.

Delete your current TFT35 (or TFT35CUR) folder off your SD card and copy fresh TFT35 from this folder: Copy to SD Card root directory to update - Unified Menu Material theme

jimhiggs, 2020-01-08:
Ok Thanks for your help. Will try that tomorrow now that you have explained the problem. I will have to find out what the difference is between the different menus.

Hukuma1, 2020-01-08:
You can see the difference on the main Github. Shows the 2 photos: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Good luck!

jimhiggs, 2020-01-08:
Thanks for all the help

guruathwal, 2020-01-19:
@jimhiggs  please close the issue if it is solved

closed 2020-01-19 by jimhiggs