bigtreetech/BIGTREETECH-TouchScreenFirmware #288: [FR] Ability to edit bed-leveling mesh grid in touch mode

Hukuma1, 2020-01-08:
**Is your feature request related to a problem? Please describe.**
[image]
Taken from: https://youtu.be/RudXmCgobso?t=224

Would love for BTT to incorporate a way to edit the mesh data that ABL creates in Marlin. UBL has this feature for Marlin mode where you can switch between the points using the knob, which definitely can be improved with touch mode I think. Ideally this would work for bilinear bed leveling, not just UBL. Would love to quickly adjust a specific mesh point with a visual guide (e.g. my left quadrant of the bed needs to be slightly lower Z offset than the rest of the bed).

**Describe the solution you'd like**
ABL/UBL manual editing of mesh z-data points after probing. But I guess if you add this code, you can also add the ability to just add the manual bed leveling mesh creation too?

guruathwal, 2020-01-16:
that looks like a 5 inch screen...

Hukuma1, 2020-01-16:
Harder to implement, for sure.

Marlin does have ULB with a similar grid layout though. That would fit:
https://youtu.be/ONpKxkil16Q?t=914

traffic-light, 2020-08-19:
I decided to make this. I hope that bigtreetech will awnsers my email about a replacement unit soon. And that I would receive one as soon as possible.

Then I can test it when building ;-)

When there is a version that's, in my opinion, good enough to be checked by others I will create a draft.

oldman4U, 2020-10-20:
Hello Hukuma1.

Your requested feature is available through PR #819 

Please help the community and close this ticket.

Thank you

Hukuma1, 2020-10-21:
Great work! Thank you all! Closing. :)

PR #1188 (2020-10-28): Mesh Editor + improvements + bugfix (replace PR #1154)
**NEW FEATURES:**
* resolves #288
* resolves #594
* resolves #777
* Implemented #288 and #777 adding a Mesh Editor (under BL menu) and Mesh Tuner menus providing a visual mesh grid (gradient colors are configurable in config.ini) and a fine tune tool for mesh editing. The Mesh Editor supports:
1) MBL
2) UBL
3) ABL Bilinear

**BUG FIXES:**
* Fixed #1167 MBL process causing "Invalid mesh" error trying to load the meshes with M420 S1. MBL process is now based on use of standard move gcode
* Fixed #967 related to wrong Software Endstops enabli [... cut]
closed 2020-10-21 by Hukuma1