bigtreetech/BIGTREETECH-TouchScreenFirmware #290: [FR] [ST7920_Simulator.c] Add a character generator in order to use klipper

derBG, 2020-01-09:
Many users of your skr boards use klipper (https://github.com/KevinOConnor/klipper).

In order to use your tft24 screen please add a character generator to TFT/src/User/API/UI/ST7920_Simulator.c

See the klipper issue on https://github.com/KevinOConnor/klipper/issues/2231

Your community will appreciate it very much

Thanks

erik23de, 2020-01-10:
I'd like to expicitely agree to the post before. Please, add a character generator!

DeviousPenguin, 2020-02-23:
I use Klipper, and an currently unable to use my TFT24 with it.

I looks like the emulation of the reprap display needs a character generator.

If you could add a character generator that would be really nice, as I'd like to use the TFT24 with the printer I am building.

DeviousPenguin, 2020-02-26:
For anyone looking to do this, the ST7920 datasheet is a good resource:

> ST7920 provides character generation ROM supporting 8192 16 x 16 character fonts and 126 8 x 16 alphanumeric
characters. It is easy to support multi languages application such as Chinese and English. Two consecutive bytes are used to specify one 16x16 character or two 8x16 half-height characters. Character codes are written into DDRAM and the corresponding fonts are mapped from CGROM or HCGROM to the display drivers.

kakou-fr, 2020-03-22:
yes please add it

AnHardt, 2020-04-02:
The original font-ROM of the ST7920 is about 130 kByte.
Even for the larger processors of the displays, this is more than 50% of the flash memory.
Better do something more useful with the memory.

JohnEdwa, 2020-04-02:
The ASCII half-height HCGROM is just 16K-bits, which is 2kB. The rest is for chinese characters, which I would assume are rarely needed with 3D printers.

bigtreetech, 2020-04-07:
> The original font-ROM of the ST7920 is about 130 kByte.
> Even for the larger processors of the displays, this is more than 50% of the flash memory.
> Better do something more useful with the memory.

BTT's touch screen has an external flash(W25Qxx) storage bitmap fonts, so don't worry about this. This function should be easy to implement, but I'm busy recently, so I don't have time to test this. After a new version is released this week, I will start to implement this function

combolek, 2020-04-24:
I would like to register another request to get this implemented. Thanks!

mbgroot, 2020-04-24:
I'm very much looking forward to a solution to this problem. The printer is without screen :(

kakou-fr, 2020-04-30:
no news ?

dushyantahuja, 2020-05-08:
This would be great.

TakeThisBanana, 2020-05-09:
I would also love to see the TFT35 E3 implemented.

Lyngtoft, 2020-05-09:
Brilliant, just switched to klipper, would love to use my tft35

bigtreetech, 2020-05-13:
[image]

[image]

Debuging... It will be realized soon

mbgroot, 2020-05-14:
the font will depend on the klipper or is it hard to set in the screen firmware?

JohnEdwa, 2020-05-14:
It's set by the screen, this whole thing is because Klipper uses the internal text generator on the display driver for the text, which was missing from this simulator. And as it is a simulator, it needs to include that character rom data as well which means it should be rather easy to swap it for a different font - if one exists or can be easily created, that is.

jdlongenecker, 2020-05-14:
@bigtreetech thank you for working on this, I know a lot of people are excited to have some high quality display options!

eboblin, 2020-05-21:
@bigtreetech any updates on this issue?

TakeThisBanana, 2020-05-21:
> [image]
> 
> [image]
> 
> Debuging... It will be realized soon

How is the debugging comming along? I am really looking forward to using my tft35 e3 again :D

JeffMedd, 2020-05-21:
I'll add a +1 of interest for this feature also! Thanks for prioritizing the development of this, it will be very much appreciated.

maketo3D, 2020-05-21:
+1 very interested on it.

RomRider, 2020-05-25:
Still a bunch of glitches, but this is great progress!!
[image]

RomRider, 2020-05-27:
@bigtreetech, when the printer starts with the latest master branch version, I have to switch to touchmode and back to marlin mode before something is displayed on the screen else it stays black.
I have `SPI_ReEnable(0);` set + the other function updated and a TFT35 v3. Also I have a SKR1.3 board.

TakeThisBanana, 2020-06-04:
> [image]
> 
> [image]
> 
> Debuging... It will be realized soon

How are things comming along?

cskozlowski, 2020-06-23:
Any update on this front?  I have a tft35 display laying around that I would really like to use with klipper on my Voron 2.4 build.

TakeThisBanana, 2020-06-26:
How is the status of developement?

GrumpaV2, 2020-06-26:
Just bought TFT 70 in big summer sale to use with Klipper on SKR 1.4.... Please implement this in firmware update

marvelous240, 2020-06-30:
Would also love to see this implemented soon!

kakou-fr, 2020-06-30:
I tried on klipper with a TFT 24 : 
1) a lot of glitch
2) 90% of the time, when the print  boot, the display only show " .... simulator"
3) when the display work, after 1-2 hour, it stop working correctly and only display the bed icon on all the screen

jariivanoff, 2020-07-26:
Adding a wish to get this fully implemented! :-D

weaslus, 2020-07-26:
yeah, this is going to be my browser homepage for the foreseeable future...

GrumpaV2, 2020-07-27:
I should have just bought Duet and Paneldue! What a waste of 65 bucks... Maybe I can hook it up to my kids Ender 3

aklef, 2020-10-06:
@bigtreetech @RomRider bump! Devs, we love you. Do you need anything? Can I help? A glass of water perhaps? We just want y’all to keep us updated please!

RomRider, 2020-10-06:
I'm not a dev here, it was just the result on my printer 😊

oldman4U, 2020-10-20:
Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

RomRider, 2020-10-20:
This isn't in any way related to #1177

oldman4U, 2020-10-20:
@RomRider 

Sorry typo #1170 not 1177.

Thank you

RomRider, 2020-10-20:
But why would you want to close this FR since it is not fully implemented? I don't understand the logic?

oldman4U, 2020-10-21:
Because there is now one centralized place instead of many tickets on a
page where nobody can find it.

RomRider, 2020-10-21:
That doesn't make any sense. Discussions on several features shouldn't happen in one mega thread it's going to be a massive mess!

Each feature should get its own discussion thread and then you can maintain your issues which references them all if you want...

oldman4U, 2020-10-21:
Are you the owner of this ticket?

There is a link to this ticket for reference FR Ticket and because it is closed does not mean that it can not be used.

But if you are the owner and you prefer a ticket on page 2 or even further behind over a pinned ticket just let me know. Such tickets like inactive tickets will be removed from the FR Ticket in about one week.

weaslus, 2020-10-21:
Are you the owner of this github?
What is your role here?

oldman4U, 2020-10-21:
I am a user like you. 

Looking at other repositories like the SKR v1.3 I try to keep this one usable.

What are you doing to help the community to make this open source project a success?

weaslus, 2020-10-21:
Im not goimg to mince words here. Appologies to those who dont care to recieve these strings of emails.

You seem like the self appointed sherrif. Your tone is tiresome. What happens if you leave the github due to boredom, busy work life, death? You want your way. Does your way really require that this feature request be closed, when it is still an active request?

oldman4U, 2020-10-21:
I will hand it over - except I am death.

In case you want to do it let me know and we can do it right now.

Again. What are you doing? I can not develop so I try to help this way and
I am sure there is also something you could do.

This ticket can remain opened no need to close it. Are you the owner?

marvelous240, 2020-10-21:
Shut up. All of you.

oldman4U, 2020-10-21:
;-)

kakou-fr, 2020-10-21:
I resolved my problems, I brought a Fysetc MINI 12864 :). It works and I can control thecolor of the screen/button  ...

DeviousPenguin, 2020-10-23:
The Fysetc is a nice workaround :) I'm doing the same, but even so it would be nice to be able to use the BTT screen as a backup or 2nd printer, I've pretty much gone Klipper only.

Stellarspace, 2020-10-28:
> ;-)

Considering it's unusual to close an issue before the problem has been fixed on GitHub, maybe you should leave a note somewhere suggesting that some issues that are closed haven't been fixed and are actually being worked on.

oldman4U, 2020-10-28:
@Stellarspace #1170 collects all FR and gives them a prominent place, instead of being almost the last one on page 4. It is also the place where future development will be organised and developers can check what's needed and if someone is already working on it.

But the owner of this ticket does not respond, so this FR will be even less important in the near future.;-(

RomRider, 2020-10-28:
You are reinventing github and how development workflows work. Good luck.
A product (open source or not) needs a Product Manager which defines a roadmap, sets priorities, etc... Apparently that's not what you do.

JeremyKennedy, 2020-10-28:
Just because the owner of this ticket doesn't respond, does not mean this ticket is less important. I am still waiting on this ticket, and I'm subscribed to it, so that any progress on it gets sent to me via email. This whole mess you've started with "cleaning up" the issues repo is messing with that... and as I'm sure you've seen nearly all responses have been negative. As @RomRider said here and I said in #1170, you are reinventing development workflows which have worked for thousands of projects so far.

Just because a repository has a large list of open/old issues doesn't mean they need to be closed. It's very likely they are still issues, and maybe the issue owner doesn't care anymore, but they should still remain open.

oldman4U, 2020-10-28:
Working in the production and broadcast industry for more than 25 years, half of the time as product manager, I am used to GIT, Atlassian Tools, on premise and cloud based support systems and so on.

Usually the owner of a repository is taking care of it, and makes decisions. Both is not the case here. An existing development workflow - I can not see it. Instead, I can see many developers leaving.

If you read my comments again you will see, that I never wrote that this ticket is less important than others - in fact, the opposite is the truth. I tried to give it a better chance to happen one day, by changing the way FR's are handled. But I almost get the feeling, that the status quo worked so well in the past, that changes are not welcome.

I can accept this.

gkatev, 2020-10-28:
Both methods are useful.
A centralized thread tracking all the FRs and their progress or what-not is helpful.
However, the individual issues should not be closed; it leads to confusion and there's no good reason to do so.

derBG, 2020-11-01:
Hello friends,

a ticket will be closed when its done. This is the normal behavior in Ticket Systems for software development. So i will close this ticket when the topic is completed.

Thanks
derBG

oldman4U, 2020-11-02:
Hi derBG.

Thank you for joining.

Of course, it is your right to close your own ticket whenever you want.
Yes, it is completely uncommon to close unsolved tickets. This would especially be true for tickets in a system, which is actively monitored and maintained by the owner, and where no alternative system is available.
It is also common to close a ticket once it is not needed any more. Many unused tickets made this repository unusable.
A Bot has been installed which will close unused tickets, even they are important.
Because of this, I added a centralised BUG list in addition to the FR list, so important Bugs are also covered.
This particular feature request is the first entry of the FR list, at the moment.
Closing all tickets which are part of the FR (and BUG) list, would help to make the ticket system more user friendly and maintainable.
Some understood and wanted to help and closed their tickets, some not.

Like it is your right to decide what you want to do with your ticket, it is also my right to decide when I spend enough time for a certain thing. And for this ticket the time is now.

Kind regards and all the best with your FR. I hope you will get it asap!

mbgroot, 2020-12-02:
Has the work on this ticket been stopped?

DeviousPenguin, 2020-12-03:
@mbgroot I think this is still being work on (hopefully)

tsndr, 2020-12-03:
I doubt it 😂

[23 more replies]

closed 2021-05-22 by bigtreetech