bigtreetech/BIGTREETECH-TouchScreenFirmware #291: Does This firmware work on the TFT35-E3 V3.0 Display?

ccgview, 2020-01-10:


Grievousbee, 2020-01-10:
From everything i read yes it should so I updated (actually think i preferred the icons that came with the card to the new ones) now I have no Marlin mode just a black screen so thinking that, no, this is not right for the TFT E3 V3 display :(
However touch screen seems to work fine. 
I say seems to as I haven't actually connected to the E3 yet.

Grievousbee, 2020-01-12:
Can confirm yes it does work.

The lack of Marlin display had nothing to do with the unit more to do with me making assumptions about my SKR1.4

dp250f, 2020-01-12:
I'm using it on mine, but if you like the encoder knob LED feature, keep your factory bin file. I've not found a way to turn those LEDs on since using this firmware.

Grievousbee, 2020-01-12:
Actually I'm not keen on those LEDs and they are the reason we have that silly extra pcb as far as I can see.

I was trying to configure it for a different printer with the SKR 1.4 rather than the SKR E3 mini but have given up on that idea and am going to fit it to my E3 now.

Chris.

Sent from MailDroid<https://goo.gl/ODgwBb>

ccgview, 2020-01-12:
Thanks for the info. Good to know

Brian Beckwith
Electronics Technologist
Canadian Coast Guard
Victoria, BC
250-413-2841<https://webmail.dfo-mpo.gc.ca/owa/UrlBlockedError.aspx>
Brian.Beckwith@dfo-mpo.gc.ca<mailto:Brian.Beckwith@dfo-mpo.gc.ca>

gl0zz3n, 2020-01-13:
> I'm using it on mine, but if you like the encoder knob LED feature, keep your factory bin file. I've not found a way to turn those LEDs on since using this firmware.

Could you please email me the factory bin? I lost mine and was trying to get my led under the knob working again. I managed to build a fw where it does indeed work, but I'm not happy with it and would rather use factory than fight to make my own. I was under the impression we were supposed to update the fw after verifying the screen was functional. 

I would really appreciate the help and can send some paypal your way to buy you a drink. 

- hairypanocha@gmail.com

dp250f, 2020-01-14:
I didn't bother to keep mine. I wrongly assumed I could download it from GitHub... That's what I get for assuming :-)

Brandis13, 2020-01-16:
Put the file "BIGTREE_TFT35_V3.0.25.1.bin" into the root directory of an SD card and put it into the TFT35-E3, the file loads to 100%, screen gets dark for a moment, then it just says "Illegal FLASH APP!"
Any idea what's wrong and how to fix?

dp250f, 2020-01-20:
> 
> 
> I'm using it on mine, but if you like the encoder knob LED feature, keep your factory bin file. I've not found a way to turn those LEDs on since using this firmware.

Did a fresh download/compile and the KNOB_LED now works - although it doesn't save the selected color, so you have to turn it on every time you reset / power off.

gl0zz3n, 2020-01-21:
> Did a fresh download/compile and the KNOB_LED now works - although it doesn't save the selected color, so you have to turn it on every time you reset / power off.

I've managed to get my led to stay on between power cycles. There's 2 ways to do this - You can either go into the led menu and find the list of colors and change led_off to the code of whatever color you want to use OR find the menu that has the init for when the device boots and change the led value there. - I forget the name of the menu off hand and I'm on my mobile. I will post a short video asap. YouTube -> Gl0zz3n Tech Reviews

alanhunter1664, 2020-05-04:
I've just updated my firmware using the BIGTREE_TFT35_V3.0_E3.26 firmware file. I included the associated TFT35 directory with the fonts and bmb files for the new menu items. However, although the firmware updates it does not update the fonts or the bitmaps and I end up with corrupt menus. Any ideas welcome please?

Grievousbee, 2020-05-05:
First suggestion is clear the sd card and just ry the fonts and bmps again on their own.

Second is try a different card. It seems the displays can be a bit finicky. Make sure format of card is correct.

Third. Were the firmware and bmp/font files actually from the same version as they wont update if not.

alanhunter1664, 2020-05-05:
Thanks, I managed to get it resolved last night and your 2nd suggestion is on the ball, it was the size and format of the SD card, I didn't read the limitations for the SD card i.e. <8G and FAT32. Once i had the right card format correctly everything worked as it should. Thank you so much for your reply.

closed 2020-08-14 by bigtreetech