bigtreetech/BIGTREETECH-TouchScreenFirmware #292: Firmware Compiler Problems.

Talha909, 2020-01-11:
I had open the platefrom.ini i had changed the line form BIGTREE_TFT35_V3_0 to BIGTREE_TFT24_V1_1.  And also changed the bed size in the configure.h. After that i compile the firmware. I got the 54 problems like that but the firmware compile successfully. After that i put the bin and folder files and in the sd card and update it. After updating all the icons are disappears.  Whats the reason behind that. 

missing braces around initializer [-Wmissing-braces]
missing braces around initializer [-Wmissing-braces]
suggest explicit braces to avoid ambiguous 'else' [-Wdangling-else]

guruathwal, 2020-01-11:
This is not a bug just ignore the warnings. you need to update the fonts and icons also.. follow the instruction here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/README.md#how-to-update-firmware

guruathwal, 2020-01-13:
@Talha909  the problem has been fixed please close the issue

closed 2020-01-14 by Talha909