bigtreetech/BIGTREETECH-TouchScreenFirmware #293: TFT35 v2.0 + encoder

Buggins77, 2020-01-11:
Good afternoon, tell me please.  I ordered Tft 35 v2.0 ... I later noticed that there is v 3.0 with an encoder and two control modes.  Can I now order a separate encoder for this board, since I saw a connector for it on version 2.0?  Or will it not work like that?  (for information - SKR 1.4 motherboard) Regards Anton

guruathwal, 2020-01-11:
no not possible. V2 does not have the required EXP connection sockets.

Buggins77, 2020-01-11:
Ok, then why there is a ENCODER input on board, beside USART3?

bkzspam, 2020-01-28:
> Ok, then why there is a ENCODER input on board, beside USART3?
> and one more question, as i have understand, with this mode, i can only send G-code command via touch screen, but what about feedback between motherboard ant TFT?
> Does some there info what possibility i can get using V2.0?

I think it's possible. Since encoder port is already available. You just need to find out Mosi, Sck and may one more pin for LCD_PINS_RS. But for full functionality you should find out more available pin. And some digging around the code. 

Of course there is feedback available. Otherwise you won't be able to get temperature (and kinda) information. For testing it simply you can send M117 gcode with a message like "M117 TFT" then a message will pop-up in the screen that says "TFT". This screen is nice, it has bugs. It should has a bit more function. But it's open source! Which is lead to me buy this screen against the MKS's screen.

bigtreetech, 2020-03-20:
Hello, LCD 12864 mode needs a hardware SPI to read the ST7920 timing sent by Marlin and parse & draw point on the TFT,  but TFT35 V2.0 no pin out hardware SPI, So usually, TFT35 V2.0 only has Touch Scren mode, no 12864 mode

closed 2020-08-17 by bigtreetech