bigtreetech/BIGTREETECH-TouchScreenFirmware #296: [FR] Idle and back to  main menu

liminghuang, 2020-01-12:
Add a new feature for configuration, idle at a function in menu for a while(ex.N minutes), system will back to main menu.

oldman4U, 2020-10-20:
Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-10-21 by liminghuang

kisslorand, 2021-01-06:
What should be considered "idle"?
No touch? No communication from TFT to Marlin? No communication from Marlin to TFT? All together? Two of them together?

oldman4U, 2021-01-11:
I guess the user has been asking for the same function which is available in Marlin, where you can define a timeout of no user action which brings you back to the main screen. Could be a timeout like it is available for dimming today, but the big question is which windows are not affected. I could imagine that windows like mesh editing, Info and so on should be excluded, but it would be good to ask the owner of this ticket.

MarkusThur, 2021-06-07:
@oldman4U 
Ok, let's discuss it a little here.

I see two "wishes" a "one touch" back to "home screen" and a "time out" that triggers it automatically, if nothing happens.

I see the "doubts" if that really make sense for every screen.

I personally would like to add the question, if the target of the "direct" return and the "time out" return should be home screen under all conditions?

I think this may should start with a "state transition diagramm" (UML).

In a first guess, I would say it should happen with the automatic backlight dimming. Alternate on a own configurable "time out".

I personally know, I regularly hate, if I leave something, as I got distracted, come back and have to go to that screen again. But some screens, definitely could autoclose if not used and go back to a screen that makes sense.

oldman4U, 2021-06-07:
The one touch is possible today by pressing the back button a bit longer.

The function this user has been asking for is available in Marlin. It can only work as long as you are on a page which is used for navigation.

But it could be a nice little project to get used to "the system".

MarkusThur, 2021-06-08:
Hmmm...
Long press already direct returns...
A timeout triggers the backlight dimming.
=>
This should be easy
If the backlight dimming would trigger the long press "event", this would be done... Think I will try it on the weekend just for fun,

A little hint where the backlight dimming and the long press return are implemented would be nice