bigtreetech/BIGTREETECH-TouchScreenFirmware #299: [BUG] second opening of internal SD results in hang on TFT

gilliands, 2020-01-14:
### Description

 Opening SKR 1.3 onboard SD for second time results in "loading..." and no further response on the TFT.
SKR 1.3 running current Marlin 2.0.1 release.  Baud 115K TFT35 E v3 or TFT24 V1.1 with touch firmware downloaded 13th Jan

### Steps to reproduce

1. power on printer. TFT establishes serial communication and sits at status screen.
2. press print icon
3. select "onboard SD" - files are listed as expected
4. press "return" icon to return to card selection menu
5. select "onboard SD" - Screen blanks with "Loading..." - no longer responds to encoder or screen input
6. TFT needs to be reset to recover control

**Expected behavior**

On second selection of onboard SD card (on the SKR1.3) the files should be displayed as they were in the first instance.

**Actual behavior**

Screen shows "Loading..." and does not respond to any further input. Repeatable on TFT24 and TFT35. 

### Hardware Variant

TFT 35 E V3 + ESP-01S(4M) or TFT24 V1.1 + ESP-12E connected to SKR 1.3 using single EXP1 (CR10 configuration) and on serial via TFT header (Reset unconnected)
### TFT Firmware Version & Main Board Firmware details

Current TFT touch firmware as of 13th Jan, Marlin 2.0.1

### Additional Information
ESP-12E was used directly on SKR 1.3 TFT header prior to using TFT display.  Worked solidly at 250K running ESP-3D.

guruathwal, 2020-01-18:
I can confirm the Bug.. it looks like something related to long file names support added through PR #235 by @melufre .

melufre, 2020-01-18:
i just do a test using an old version branched version of the 06 of december ( before the bug of squared icons corrected) and i do not reproduced the issue , in next days i will merge my branch with the official one and do another test

Grayzone233, 2020-01-20:
I confirmed the bug as well. Also if you go ahead and select a file for print, then cancel the print you also end up with ----Screen blanks with "Loading..."

melufre, 2020-01-23:
So it seems error occurs on requestCommandInfo.cmd_rev_buf = malloc(CMD_MAX_REV);
If i remove the long file name support the error is less systematic but it occurs after some tries.
It seem also that there is not missed clearRequestCommandInfo() calls i check it using a counter .
so i test by making static (no more malloc) requestCommandInfo and it do not reproduce issue.
typedef struct {
    char command[CMD_MAX_CHAR];     // The command sent to printer
    char startMagic[CMD_MAX_CHAR];  // The magic to identify the start
    char stopMagic[CMD_MAX_CHAR];   // The magic to identify the stop
    char errorMagic[CMD_MAX_CHAR];  // The magic to identify the error response
    bool inResponse;                // true if between start and stop magic
    bool inWaitResponse;            // true if waiting for start magic
    bool done;                      // true if command is executed and response is received
    bool inError;                   // true if error response
    char cmd_rev_buf[CMD_MAX_REV];              // buffer where store the command response
} REQUEST_COMMAND_INFO;

gilliands, 2020-01-24:
Might there be an issue with the way malloc is working/being used in that is isn't freeing up memory properly which might account for other people's issues with things working and then stopping some time later? If something is being freed up but referenced then writes can end up going anywhere. (#283, 303, 306)

gilliands, 2020-01-24:
There seems to be a lot of free() in the code.... This is an embedded system and there is unlikely to be any heap collection so is there any way we can allocate RAM *ONCE* and not free it otherwise we run out of heap rather quickly....  Thoughts?  Any large buffers need to be the maximal size as we can't extend and if we free them, we loose the buffer RAM and allocate fresh from the top of the heap...until we fall of it and *BANG*.

melufre, 2020-01-24:
more over requestCommandInfo.cmd_rev_buf  is 5000 bytes so it could fragment whole RAM quickly .
and after some time there is no space to allocate 5k
 i have the feeling that something has been added in the last 1.5 month that make more dyn RAM consuption.

do you know if there is a way in C to get the heap health  ?
do you test the char cmd_rev_buf[CMD_MAX_REV];  static definition ?

gilliands, 2020-01-25:
Not tested yet but I can see how it would fix this issue. The easy way (for the coder) would be to test by mallocking a 32 bit word and check the address pointer against the known end of the stack then free it immediately so the heap shrinks back without fragmentation.  The other way (for the hardware engineer) is to break out the SWD debugger and watch what happens to the unmodified code.   Goes off looking for my debugger....
I can see why malloc is useful but not if buffers are freed and re-created when they may no longer be at the top of the heap. Why not create the pointer to the malloc structure (zero by default) and if zero, create it - if not zero then use it.  Freeing would be zeroing the area of a non zero pointer but leave the memory allocated.  Not as elegant but for an STM32, very memory efficient.

melufre, 2020-02-04:
i have test this patch on in fork and it work very well . here is the patch details
```
....
    char cmd_rev_buf[CMD_MAX_REV];              // buffer where store the command response
    //char *cmd_rev_buf;              // buffer where store the command response
} REQUEST_COMMAND_INFO;

static void resetRequestCommandInfo(void) 
{
  //requestCommandInfo.cmd_rev_buf = malloc(CMD_MAX_REV);
  //while(!requestCommandInfo.cmd_rev_buf); // malloc failed
....

void clearRequestCommandInfo(void) 
{
  //free(requestCommandInfo.cmd_rev_buf);
}
```

regarding select a file for print, then cancel the print you also end up with ----Screen blanks with "Loading..." 

 i have also dig until the fact that when the print is start(after M24) marlin don't whant to answser to commands even M105 .
this happen also when the print is started from Marlin but only on the TFT serial not the USB one

here is my Marlin cfg
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

so if someone knows what to configure on Marlin your are welcome

gilliands, 2020-02-04:
While holding my copy of K&R C (second edition) in one hand, I've been looking at the inappropriate use of malloc() in the software where memory is claimed, used and freed out of order which ends up fragmenting and exhausting the heap.  You've done a much better job than me in replacing the filesystem stuff for static allocations :-)  I think one of the other areas is with the serial buffer allocation where each time an interface is brought up and down, memory is malloced and freed before being claimed again...

gilliands, 2020-02-05:
Melufre's patch is a big improvement on normal behaviour.  I think we're on the right track in eliminating the inappropriate use of malloc() which is causing heap fragmentation and exhaustion.  There are tests to see if the malloc is working but no attempt to report an error if any fail.

closed 2020-08-17 by bigtreetech