bigtreetech/BIGTREETECH-TouchScreenFirmware #304: [BUG] TFT35 v3/Relay 1.2 : shutdown function

arut16, 2020-01-17:
### Description

Hi,

I ordered and installed your SKR 1.3, yours TFT35 v3 (version 3.0.25) and yours Relay 1.2 from your e-store on AliExpress to my ender 3 Pro.

All seems to working well instead of the auto shutdown function from the couple TFT35 V3+ RELAY 1.2. I'm, of course, activated auto power function in options menu. 

I can't find what's I'm doing wrong...

My wiring between them : 
[Relay][S] <=> [TFT35][PA12]
[Relay][G] <=> [TFT35][GND]

When I plug 230v cable, relay is closed (green led lights) and all is powered BUT approximately 10s later without any interaction, relay opening by it self (green led get off and a red one lights instead) and cut the power to PSU... 

I figured it out that if I plug a button (like stock end stops on ender 3) on reset and restart pins on the relay, the two LEDs green and red lights, relay close again and the 230v get to PCU to power printer but M81 command at the end gcode not working... Screen showing a message to cool down hot end, then "shutting down" but noting append... Printer remains ON...

Can you please help me ?

### Steps to reproduce

1. Connecting TFT35 V3 to Relay 1.2 from BTT
[Relay][S] <=> [TFT35][PA12]
[Relay][G] <=> [TFT35][GND]

2. Print something to send M81 command

**Expected behavior**
Printer shutting down

**Actual behavior**
Shutting down after 10s or never...

### Hardware Variant

BTT TFT35 V3 with 3.0.25 firmware. SKR 1.3 with Marlin 2.0.x bugfix

emtrax-ltd, 2020-01-19:
have the same problem. i have tested a lot of combinations and figured out that the pin high voltage (~0.2v - 0.3v) of ps-on of the tft35 is not enough to keep the relay alive. test -> if you wire the 3.3v of the ps-on to the signal line then the relay keeps alive... until you remove it again -> then after 10-15secs the relay switches off.

emtrax-ltd, 2020-01-19:
just another bug -> the ps-on was mislabeled ("pa12")  in the pin-configuration until 3.0.25. this was fixed in 3.0.25.1 -> now it's "pc12".

emtrax-ltd, 2020-01-19:
ok,... i've got it now! the unexpected bahavior is based on the wrong pin-configuration in variants.h of the tft-firmware -> was noit fixed in 3.0.25.1.

just change

#define PS_ON_PIN      PA12

into 

#define PS_ON_PIN      PC12

and everthing is fine. now you only need to wire PS-ON to the G/S on the relay.

arut16, 2020-01-19:
> ok,... i've got it now! the unexpected bahavior is based on the wrong pin-configuration in variants.h of the tft-firmware -> was noit fixed in 3.0.25.1.
>
> just change
>
> #define PS_ON_PIN PA12
>
> into [...]

Thanks for your investigation emtrax-ltd ! I will try that. Will feedback when here. ;-)

arut16, 2020-01-21:
> ok,... i've got it now! the unexpected bahavior is based on the wrong pin-configuration in variants.h of the tft-firmware -> was noit fixed in 3.0.25.1.
>
> just change
>
> #define PS_ON_PIN PA12
>
> into [...]

Bingo ! You are right ! This is the solution. All working propely now. Command M81 shuting down printer and relay stay closed even after 10 sec. I thing TFT screen is sending always the signal to the relay but then you send the command M81, signal from TFT stops and the relay opening after 10sec as it programmed.

Perfect !!

Thank you VERY much !

closed 2020-01-21 by arut16

dziabi3, 2020-02-03:
I see that the error has already been reported ;-)
I came across this error in software 25.1 for TFT35 E3 V3.0.