bigtreetech/BIGTREETECH-TouchScreenFirmware #306: [BUG] KNOB_LED color doesn't save (BTT-TFT35-E3-V3.0)

dp250f, 2020-01-19:
### Description
When I select a color for KNOB_LED on my BTT-TFT35-E3-V3.0, it does not stay that color on next startup.

### Steps to reproduce
1. Power on BTT-TFT35-E3-V3.0 and enter Touch Mode.
2. Select Rotary Knob LED color in Feature menu and go back to main menu screen.
3. Press reset button and notice the KNOB LED does not return to the color you selected.

**Expected behavior**
KNOB_LED should return to the color you select in the Feature menu.

**Actual behavior**
KNOB_LED turns off after reset or power on. Does not return to selected color.

### Hardware Variant
BTT-TFT35-E3-V3.0
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0.25.1 (commit 5777f41)
Marlin 2.0.1 (commit 0673f33)

### Additional Information
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4083233/Configuration.zip)

wallen83, 2020-01-28:
The same for me. I checked in atom if I could find it in the fw, but found nothing. 
I can't even find where to change the color of the knob in the menu. Found it once, but not now

wallen83, 2020-01-28:
I found a solution!  Go in to atom and and open the firmware 
go to tft,src,user,meny  and open file ledcolor.h    row7 #define LED_OFF                   0x00000000 
enter the color code you want there, then it will start up with it     in the picture I have entered red code 0x0000FF00  = red 

[image]

cht47, 2020-02-05:
I confirm the bug. To hardcode the color is one solution but it should save like it did with the older firmware.
By the way if you switch between Marlin mode and TFT mode it also resets the LED.

jscheiben, 2020-02-14:
Just got my board and can also confirm the issue. I've hard coded the color but that defeats the purpose of having the option to chose a color.

jpaudio, 2020-02-15:
i have same problem with knob led light not saving

GeminiServer, 2020-02-29:
https://github.com/GeminiServer/BIGTREETECH-TouchScreenFirmware

Feature:
Knop LED set/save/restore settings support. 
And some other little improvements. 

Master is for all

Release is my private adaptions.

Testmangh, 2020-03-01:
Same problem here!

dp250f, 2020-04-01:
I compiled a fresh firmware today and this is now working. Thanks!

closed 2020-04-01 by dp250f

dineshvyas, 2020-05-03:
This is the most stupid question but can someone tell me what is KNOB LED?
BTT TFT35 V3 my screen has a Knob but no led.

jpaudio, 2020-05-03:
do you go into settings you can turn the LED on you can make it whatever color you want and now it's new firmware you can actually save it so when it turns on it stays that colorDJ JPA

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/306#issuecomment-623082888",
"url": "https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/306#issuecomment-623082888",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]