bigtreetech/BIGTREETECH-TouchScreenFirmware #307: [BUG] TFT35 v3.0 Slowly Stops Working 5-8Mins After Boot FW V25

pokecambruh, 2020-01-20:
#245 # Description

TFT35 v3.0 configuration has an ESP32S, connected to a BTT Mini E3 1.2 MB running Marlin 2.0.x Bugfix, and a BLtouch v3.1. The TFT35 itself works perfectly for about 5 minutes, all functions work correctly and can control all printer functions/display menus and values correctly. After about the 5 minute mark, the display will go back to the OS option menu (Emulation or Touchscreen). Select whichever one, it continues to work for about 2 minutes, then back to the OS menu. Again, select, works for ~1 minute, back to OS menu. Again, ~30 seconds, menu. Repeats until it only flashes your relevant OS screen for a second and then permanently stays at the OS menu, unable to select or go anywhere from there.

The remedy I've found is letting the display sit without power for ~30 minutes, come back and it works again to repeat the same cycle. I imagined overheating could play a part, but with a thermal gun, the entire

### Steps to reproduce

1. Turn on printer (direct USB power or mains power from printer PSU, doesn't matter)
2. Let sit powered on for 5-8 minutes
3. Screen becomes unresponsive and unusable, permanently stuck at OS menu

**Expected behavior**
Printer should work at all times

**Actual behavior**
TFT35 screen sporadically, slowly reverts back to OS option menu during use until it is permanently stuck there and becomes unresponsive.

### Hardware Variant
TFT35-E3 v3.0

### TFT Firmware Version & Main Board Firmware details
v25, v25.1 (TFT35-E3 v3.0), v2.0.x Bugfix (BTT Mini E3 1.2)

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

guruathwal, 2020-01-31:
your touch screen is sensing false touch. If you put your TFT in a case, try without any case.

dziabi3, 2020-02-20:
How I dealt with this problem by removing the display from the steel frame.
I also disconnected the reset signal from the tft tape (rs232).

Scope666, 2020-02-20:
I would consider that a pretty major design flaw considering the E3 is DESIGNED to go in the stock metal frame of the Ender 3.

hfog, 2020-02-20:
@Scope666, I won't disagree. I dealt with the problem by slipping washers onto the screws so that the board isn't pressing right up against the case.

closed 2020-08-14 by bigtreetech