bigtreetech/BIGTREETECH-TouchScreenFirmware #308: [FIX] TOUCH SCREEN miss calibrated (reset and calibrate new)

ORGATHM, 2020-01-20:
After flashing new firmware, i saw only pixel blocks, no characters.
so i touched not the red dots and miss configured the touchscreen.
it was not possible to enter the right menus. flashing all versions of firmware does not 
got me the option to calibrate again.

FIX:
after reading the source code i found a hint, that i have to create a new file 
named: **reset.txt**
**copied to the sd card, restarting(reset) display.** 
now i was able to do a new calibration.

maybe you have to insert this info into your manual. not everybody can read and understand c++ source code.

ddggrr, 2020-05-23:
I have the same problem.  No menu on startup after flashing firmware. Only this screen. 
[image]
Help me please.

jotavcd, 2020-07-22:
im having the same problem. what did you put inside the reset.txt?

closed 2020-08-17 by bigtreetech