bigtreetech/BIGTREETECH-TouchScreenFirmware #312: Smart Filament Sensor dosen't work

Capricornus73, 2020-01-25:
The smart filament sensor won't work in touch screen mode. The sensor ist conected in according to the TFT35V3 / Filament detection module manual directly at the TFT 35v3.0. At the feature menue the filament runout! is toggled to smart. What must i do to enable the sensor in the TFT35 v3.0 Firmware?

Thank you for helping me

arut16, 2020-01-25:
Same issue here. 

I succed to make it work properly to connect it directely to SRK 1.3 (+X socket) board AND only trough Marlin mode AND under condition to send command "M412 D7" with PrinRun when M500 to insert the 7mm offset who is not working if coded in Marlin code on SKR directely.

Any suggestions ?

My config :
Ender 3 Pro
SKR 1.3 with Marlin 2.0.x bugfix
TFT35 v3 with 3.0.25.1

Capricornus73, 2020-01-26:
My board is skr pro v1.1 with tft 35v3.0 and my printer is JGAurora A5.
Directly connected to the Board is the original filament sensor and it works fine in Marlin Mode.
I want to use the smart filament sensor with the tft 35  in touchscreen mode, but it dosen't work!

guruathwal, 2020-01-26:
did you tried changing the Filament runout option in feature setting in the TFT menu to `Smart`?

Capricornus73, 2020-01-26:
Yes i did, but it won't work

phtp, 2020-01-30:
It doesn't work for me either. I ended up just directly connecting to the SKR Mini E3 v1.2 board.

arut16, 2020-01-30:
Is anyone succed to connect the smart filament sensor to a must be compatible TFT screen from BTT AND make it work ??

Scope666, 2020-01-31:
I just tested a non-smart sensor connected in the FIL-DET port of the TFT35 V3  and it doesn't work either.  I have it set to "on" in the menu.

guruathwal, 2020-01-31:
Did you guys tried setting this line to false?
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/5777f41b5f8c1a41410a6874614499c91ac78fa2/TFT/src/User/Configuration.h#L119

Scope666, 2020-01-31:
I haven't yet (will try tomorrow) but I did repeatedly insert and remove a piece of filament during a print.  I would think even with it reversed, the port would have detected a change on either the insertion or removal ... no?

guruathwal, 2020-01-31:
I can't confirm it.
but if assumed that the sensor only send a HIGH signal when break is detected and send no signal (neither high or low) at all when no breakage then input pin on TFT if set to high will always read it as high? need testing to confirm this.

guruathwal, 2020-01-31:
Also the manual for the smart filament sensor states it support the TFT24 & TFT35V3 in 12864 mode.

Only @bigtreetech can comfirm this :

https://github.com/bigtreetech/smart-filament-detection-module/blob/master/manual/smart%20filament%20sensor%20module%20manual191205.pdf
[image]

guruathwal, 2020-01-31:
confirm this also if this was enabled in your firmware..
according to Bigtreetech manual following line also needs to be un-commented this line in marlin firmware: 
`#define M114_DETAIL`

https://github.com/MarlinFirmware/Marlin/blob/8eb6603e2aa948f54f1e6faec264f7309b214a89/Marlin/Configuration_adv.h#L280

[image]

arut16, 2020-01-31:
> confirm this also if this was enabled in your firmware..
> according to Bigtreetech manual following line also needs to be un-commented this line in marlin firmware:
> `#define M114_DETAIL` [...]

Thanks for your help. Will check again all tonight, flash firmewares and try again to trigger this feature...

Scope666, 2020-01-31:
> I can't confirm it.
> but if assumed that the sensor only send a HIGH signal when break is detected and send no signal (neither high or low) at all when no breakage then input pin on TFT if set to high will always read it as high? need testing to confirm this.

So I did some extensive testing.  With FIL_RUNOUT_INVERTING set to true, no change when printing when inserting or removing filament from sensor.  LED on sensor is dim when no filament, bright when filament present.

With FIL_RUNOUT_INVERTING set to false, there IS a change depending on if filament is present or now, BUT it is inverted.  If filament is in sensor, printing pauses and "filament runout" pops up on screen.  Can only resume by removing filament from sensor.  LED on sensor is OFF when no filament and bright when present.

I believe I'm going to have to solder the sensor itself to  change it's state from normally closed to normally open.  Before I do I wanted to see if you agree @guruathwal ?

guruathwal, 2020-01-31:
@Scope666 try this:
change the line number 40 in extend.c from:
`  GPIO_InitSet(FIL_RUNOUT_PIN, FIL_RUNOUT_INVERTING ? MGPIO_MODE_IPD : MGPIO_MODE_IPU, 0);`
to
`  GPIO_InitSet(FIL_RUNOUT_PIN, MGPIO_MODE_IPU, 0);`

then try with FIL_RUNOUT_INVERTING set to both true and false.

Scope666, 2020-01-31:
> @Scope666 try this:
> change the line number 40 in extend.c from:
> ` GPIO_InitSet(FIL_RUNOUT_PIN, FIL_RUNOUT_INVERTING ? MGPIO_MODE_IPD : MGPIO_MODE_IPU, 0);`
> to
> ` GPIO_InitSet(FIL_RUNOUT_PIN, MGPIO_MODE_IPU, 0);`
> 
> then try with FIL_RUNOUT_INVERTING set to both true and false.

Thanks for this, I will try this shortly and report back.  I'm glad I waited to solder.    :)

arut16, 2020-01-31:
> @Scope666 try this:
> change the line number 40 in extend.c from:
> ` GPIO_InitSet(FIL_RUNOUT_PIN, FIL_RUNOUT_INVERTING ? MGPIO_MODE_IPD : MGPIO_MODE_IPU, 0);`
> to
> ` GPIO_InitSet(FIL_RUNOUT_PIN, MGPIO_MODE_IPU, 0);`
> 
> then try with FIL_RUNOUT_INVERTING set to both true and false.

Will try that when my actual printing will finish with my Smart Filament Sensor and report back here.

guruathwal, 2020-02-01:
@Scope666  good to know that it works.. it still needs confirmation for the smart sensor also

arut16, 2020-02-01:
> @Scope666 good to know that it works.. it still needs confirmation for the smart sensor also

I'm back ! Sorry for feedback time... kids...
I tested with the modification in extend.c (line 40) with FIL_RUNOUT_INVERTING on "true" and "false".

In all cases "smart" is selected in touch screen menu. I made tests in touch mode and marlin simulation mode.

Unfortanately, in none cases I'm not getting any reaction from printer when I'm cutting the filament.

Well, not really none any reaction in marlin simulation mode : in both situation ("true" or "false") with filament intact, when I'm starting a print, after the first 7 mm, fil runout/pause appears on black/green screen : "hotend warming" when filament is get out when an alarm is triggered and a message saye to me to change the filament. When I'm push rotary knob, filament is purged. When purging is done I have two possibilities "purge more" or "filament runout detection : on". I'm selecting the second one. Hotend warms again and when printing continue. BUT after that if I cut the filament, nothing append... printer continue to "print"...

Scope666, 2020-02-01:
Before I put guruathwal's code edit in, I had some strange behavior also, a few times it would completely skip my starting gcode.  (auto home, draw the purge line)

Seems better now with the code change.

guruathwal, 2020-02-02:
@arut16  the smart sensor also requires `M114_DETAIL` enabled in marlin firmware.
did you enabled that ?

arut16, 2020-02-02:
@guruathwal Yes M114_DETAIL is uncommented.
[image]

I made a little video also :https://youtu.be/DYPvalHALLA

arut16, 2020-02-03:
@guruathwal any ideas ? 🤞🙏

guruathwal, 2020-02-04:
@arut16 its hard to diagnose the issue without the actual devices. BTT was sending me a smart sensor but they are still on holidays due to the coronavirus outbreak in china.

can you share the finding **without** changing line number 40 in extend.c from:
`GPIO_InitSet(FIL_RUNOUT_PIN, FIL_RUNOUT_INVERTING ? MGPIO_MODE_IPD : MGPIO_MODE_IPU, 0);` 

and just with the `FIL_RUNOUT_INVERTING` set to both true and false.

arut16, 2020-02-04:
@guruathwal I understand. I have same problem  with an BMG Extruder order : still not sending.

Will restore line 40 in extend.c and make tests with "true" or "false" value. I will share findings here. 

Thanks !

arut16, 2020-02-04:
@guruathwal Unfortunately no configuration works ... I tried with the "true" and "false" parameter, in touch mode and Marlin mode. But nothing to do ...
Below the settings activated according to the manual of the smart filament sensor:
[image]
[image]
[image]
[image]
[image]
[image]

[Here is the link to the source code of Marlin 2.0.x bugfix currently installed on my SKR 1.3 with all my configurations.](https://1drv.ms/u/s!AtRfZZCsfGrj-B3F1YtdxiKV7tiW?e=CBNgcr) 

The TFT35 v3 firmware is your last release (3.0.25.2) apart from the following change which corrects an error in the BTT source code for BTT 1.2 Relay (PA12=>PC12). It's working, you can add this fix in your next update of your own release. ;-)
[image]
[image]

guruathwal, 2020-02-05:
@arut16 ... 🧐 So changing the Power Supply pin number from PA12 to PC12 makes the filament runout sensor work which is on PA15. Am I right?

arut16, 2020-02-05:
@guruathwal No. It's a little comment about BTT 1.2 Relay https://a.aliexpress.com/_U37BR
The problem with smart filament sensor is still actual

guruathwal, 2020-02-05:
@arut16 ok... For a moment I thought the PIN numbers are really messed up 😂.
Try this - change line number 40 in extend.c to:
`GPIO_InitSet(FIL_RUNOUT_PIN, MGPIO_MODE_IPD, 0);`
**this just means to set the pin to Input Pull Down earlier suggestion was to set to Input Pull Up which worked for @Scope666**

Then try with `FIL_RUNOUT_INVERTING` set to both true and false.

arut16, 2020-02-05:
@guruathwal Unfortunately i have already tried and the tests are inconclusive.  See above. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/312#issuecomment-581067265

guruathwal, 2020-02-05:
@arut16 there are two possible changes to line No 40:
The first suggestion was:
`GPIO_InitSet(FIL_RUNOUT_PIN, MGPIO_MODE_IPU, 0);`

The second suggestion now is
`GPIO_InitSet(FIL_RUNOUT_PIN, MGPIO_MODE_IPD, 0);`

The 2nd variable name is different in both of these **_IPU** & **_IPD**
did you already tried these? In combination with FIL_RUNOUT_INVERTING?

arut16, 2020-02-05:
Mea Culpa ! I didn't noticed the difference. I didn't tried the second one. Will try and back here.

arut16, 2020-02-05:
@guruathwal Tests done. Still not triggering filament runout...

guruathwal, 2020-02-06:
did you received any datasheet / schematic with the sensor? I need the datasheet to know what it actually does.

arut16, 2020-02-06:
@guruathwal Unfortunately not. Only the sensor.  Will to disassemble the sensor to see what inside if it can help us.

guruathwal, 2020-02-06:
@arut16  do it only if you know what you are doing. if you are willing to do it then share the PCB layout as clear as possible.

carl1961, 2020-02-06:
I already tore mine apart months ago to learn. The smart sensor is like a optical sensor, except Marlin treats it like a encoder. It also wires up like a optical endstop.  First check it is working by M119 and see what it reads then using a small stick or screwdriver barely turn the wheel through the window and the M119 should change . The ticks on the wheel or very close so it is easy to skip and not see a change. Also when using the smart sensor if you hotend is to close to the bed and not letting the filament move enough it will trigger a out of filament. Here is link to my firmware with smart sensor working in emulator screen. https://drive.google.com/file/d/1IYu08tTJ0wdAVhK9Cpacb6pjYZDcMeN8/view?usp=drivesdk

arut16, 2020-02-06:
@carl1961 Thanks for your feedback. Will try to do some tests with M119.
Do you succed to make it work on touch mode ? Do you connected sensor on your main board or on TFT ?

carl1961, 2020-02-06:
Mine's connected to my xmax end stop, skr 1.1 pro

carl1961, 2020-02-06:
I haven't tried it on a touchscreen side just in simulator

arut16, 2020-02-06:
@carl1961 I succed also to make it work by connecting directly to SKR 1.3 on "X+" pin and only on Marlin/simulation mode. 

The purpose of this issue is to find out why it doesn't work connected to TFT35 v3 as it can be if you follow sensor's btt's manual...

carl1961, 2020-02-06:
in the manual, and in this video
https://www.youtube.com/watch?v=iFGIKMAGJTk    it shows it has to be
connected to the mother board . I think it needs to  utilize the marlin
firmware  to be able to allow the  pulsing (on/off)of the filament switch
to work as a encoder at which the firmware for the touch screen does not
have this code to deal with the pulsing of the smart sensor.   manual also
says at #5  5) The module is being optimized and will be compatible with
touch
screen in the future. the code in marlin basically counts the pulses and is
no pulses happen for the default 7mm it triggers a out of filament or jam.
so I think it best to just use your mother board until BigTreeTech adds the
code to there touchscreen. from what I see it seems the touch screen is
it's own controller with it's own software for the basic stuff (Like the
MKS TFT does. but the simulator side is using all marlin firmware
<https://www.youtube.com/watch?v=iFGIKMAGJTk>

arut16, 2020-02-06:
@carl1961 Ok. Your spitch makes sense and I partially understand. BUT how to interpret figure 14 (page 12) of the manual and the connector which is present on the touch screen? If the sensor must ONLY be connected to the motherboard to work, why is there this connector? For future software development?

[image]

[image]

arut16, 2020-02-06:
@carl1961 @guruathwal I made few tests with M119 with Proterface (`FIL_RUNOUT_INVERTING` on `true` and line 40 in `extend.c` untouched) and "filament" in triggering but I can't get any reaction from touch mode interface. In Marlin mode, working good as expected.

It's really a shame to have such a beautiful color touch screen and be forced to use Marlin mode to see this thing work properly ...

[image]

carl1961, 2020-02-07:
@arut16 I saw that picture too, they need to make it clear as it is contradicting the manual, I see
 it as remove from there and put on the board. most likely a regular filament sensor may work there.

guruathwal, 2020-02-07:
so the sensor does not contain any fancy micro-controller. I have uploaded a test firmware for the TFT35-V3 to debug and check if it is detecting the pulses properly from the smart sensor. 

https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware-test-firmware-binaries/tree/master/Filament%20sensor%20test%20binary

you need to flash it to your TFT screen (**make sure you device is TFT35-V3**) and follow these steps:

1. disconnect the TFT from the main board and just power on the TFT.
2. connect the smart sensor to the TFT.
3. change `Filament Runout Sensor` to `Smart` in setting.
3. Open G-Code Terminal in your TFT and move the filament up / down in the sensor manually .
4. if the TFT is able to sense the pulses form the sensor then it will print text like this: `Sensor Level: 1` based on the HIGH or LOW state of the input pin.  if it does not show in the gcode terminal in TFT then try connecting it to the PC via USB-Serial converter.
5. share your finding.

**do not use this firmware for printing**

arut16, 2020-02-07:
@guruathwal Ok. After just powering only TFT and connecting sensor, I did not succed to get any extra lines except M105 and M114 E when "Smart" is selected...

There the demostration video  : https://youtu.be/hG59woCCwBg

Unfortunately I haven't any USB-Serial converter to use for connecting TFT to PC. I even don't know how do it. Sorry my knowleadge is not so extended as yours.

What do you thinck ?

guruathwal, 2020-02-08:
@arut16 can you try it again with BTT-WRITER-V1.0 and keeping the TFT on Status Screen.
Then check it on the PC

arut16, 2020-02-08:
@guruathwal You were right. The sensor effectively regenerates the pulses received by screen and retransmetted via serial port... The issue is maybe on interpretation of Marlin on mainboard of this pulses...

=> https://youtu.be/KmSxT3bPWYY

guruathwal, 2020-02-09:
@arut16 .. great... Now can you test it with actual printing test?

arut16, 2020-02-09:
@guruathwal What do you mean ? In same conditions (isolated from main board, connected to the sensor and putty), start a test print ?

guruathwal, 2020-02-09:
@arut16  no.. connect the TFT to the main board and print an actual test print with the sensor connected to the TFT then try to block the filament after fer minutes.

arut16, 2020-02-09:
@guruathwal with actual testing firmeware on TFT ?

guruathwal, 2020-02-09:
@arut16 yes with the test firmware I uploaded
https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware-test-firmware-binaries/tree/master/Filament%20sensor%20test%20binary

arut16, 2020-02-09:
@guruathwal Tests done. Well it's strange... During printing, I stopped the filament by hand but "sensor filter level: 0 or 1" never appeared in the data flow but only a single occurrence when I canceled the printing in progress (photos below) :

[First video WITHOUT BBT Writer](https://youtu.be/JXeLn2l2Qek)
[Second video with capture on Putty](https://youtu.be/GQG6KC1PfyY)

[image]
[image]

guruathwal, 2020-02-09:
@arut16 no. Just print normally without BTT writer (and ignore the unknown command warning).. you do not have to look at serial data.. you need to test the filament runout sensor. Just see if it detects the filament when blocked while printing.

arut16, 2020-02-09:
@guruathwal I understood you.  I also tried without BTT writer: no reaction. Video : https://youtu.be/JXeLn2l2Qek

guruathwal, 2020-02-09:
You did not print anything.. you can not test the sensor without printing.

arut16, 2020-02-09:
@guruathwal sorry. I didn't make video but I maked a test just after this first test and it not worked.

guruathwal, 2020-02-09:
Ok I got it.. will take a look at it tomorrow..

[64 more replies]

closed 2020-11-25 by stale[bot]