bigtreetech/BIGTREETECH-TouchScreenFirmware #313: [Issue] TFT35 Previews not showing up correctly.

bkzspam, 2020-01-25:
Hi all,
I have a TFT35 V2 with latest firmware configured by self. My problem is previews not showing up or just a "static" picture. This is happening when printing files. 
For preview I'm creating a picture 95x95 with this name format: file_95.bmp

Pictures:
https://pasteboard.co/IREDyOU.jpg
https://pasteboard.co/IREE3hJ.jpg

Hukuma1, 2020-01-26:
Are you able to create custom icon just fine? Could be an issue of how you're saving the BMP file.

bkzspam, 2020-01-26:
> Are you able to create custom a icon just fine? Could be an issue of how you're saving the BMP file.
> 
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/readme/English/How%20to%20custom%20icon%202019-11-11.pdf

I already made some icons, they are working fine. Only preview not works.

rndsum, 2020-02-12:
Copy it to the SD card again and let it refresh, if that doesn't fix it re-export it so it rewrites the main image. 

It's a sign that the file was corrupted during one of the processes either writing to the SD card on the computer/display being dirty, or the file from whatever photo editor your using.  I've had the issue with several things best to write the images one at a time to the firmware if you can.

Might also want to wipe down your card with some alcohol & let it dry to ensure there's no oils on it.

alagerqvist, 2020-02-14:
I took one of the bmp files provided in the github for firmware and renamed it to reflect the <file>.gcode => <file>_95.bmp and it worked just fine when started print.
I have a TFT35 V3 but that should not make any difference in this case.
I think you have the wrong format of the bmp file, there are many different ways to make a bmp and the firmware is somewhat restricted.

bkzspam, 2020-03-08:
I used modelbuilder but nothing changed. Just once it worked out well. I still don't know why this is happening.

oldman4U, 2020-07-21:
Hi. BTT released a great way to get this function today. Could you please check and close this ticket afterwards. Thank you

closed 2020-08-14 by bigtreetech