bigtreetech/BIGTREETECH-TouchScreenFirmware #314: Status Line at Top not actual at printing and at print end the Display stays on the Print status

Powerprobot, 2020-01-26:
I used the Unified Menu at TFT35 1.2

While printing the Status Line in Top of the TFT does not actualize the temperature from the Nozzle an Bed.

If i go in the "more" options at printing, the Status line will actualize the temperature.

Another Bug:
If the print finish the display stay on Print-Status. There is no popup to tell the finishing print.
.... and there is a Bug to decode the G-Files...
My printer give an Error.

[image]

emtrax-ltd, 2020-01-27:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/499e6f3533db9d9bde24d4c1dae92ef7b4268889

mikefromups, 2020-02-25:
For version TFT Firmware version 3.0.25.1.

Remove Global Extruder and Bed temp from "Printing" Screen so it doesn't cover up your filename.  It is not needed on this screen since it is already in the center.

There is a new file in 3.0.25.1

TFT\src\User\Menu\Parametersetting.c

Change this line 
if(infoMenu.menu[infoMenu.cur] == menuMove || infoMenu.menu[infoMenu.cur] == menuStatus) return;

To This
if(infoMenu.menu[infoMenu.cur] == menuMove || infoMenu.menu[infoMenu.cur] == menuStatus || infoMenu.menu[infoMenu.cur] == menuPrinting) return;

closed 2020-08-17 by bigtreetech