bigtreetech/BIGTREETECH-TouchScreenFirmware #315: beeper stuck

Chad541121, 2020-01-26:
When switching from marlin to touch mode, the beeper stays on.  I have to shut down machine to make it stop

hfog, 2020-01-31:
I have encountered the same thing. Turning off the beeps in the touch menu is a workaround but obviously not an ideal solution.

cht47, 2020-02-11:
Confirm. I don't exactly what setting it triggers, just swapping from Malin to Touch Mode doesn't trigger constant beep, but I did a manual bed level, stored settings and switched back to touch and beep.

hfog, 2020-02-14:
I'm pretty sure it's an alarm that occurs if you try to switch during a print that was started from an external source (in my case Octoprint). I've been working on changes to the TFT firmware that allow it to continue to poll UART activity even when in emulation mode and since I've been testing that I haven't encountered this issue again.

hfog, 2020-02-21:
After quite a bit of experimentation and looking through the firmware to fix other issues, I think that this problem is caused by the firmware not tracking temps when in Marlin emulation mode. The printer is tracking temps and the info you're seeing in emulation mode is what the printer has been tracking; the TFT is not tracking anything at all, just displaying whatever image the printer is telling it to display. When you switch back to touch mode, the TFT suddenly starts getting temperatures and if you've been printing, they'll have suddenly jumped which puts the TFT into an error mode.
Since I've been running my own version of the firmware which monitors serial traffic routed through the TFT no matter which mode you're in (since merged into guruathwal's master branch for the next release https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware) I have only encountered this behavior once. That one time was when my thermistor was unplugged (my doing), correctly triggering a thermal runaway error.

closed 2020-08-14 by bigtreetech