bigtreetech/BIGTREETECH-TouchScreenFirmware #316: [Q] How to disable top status line

billydragon, 2020-01-27:
Can anybody can tell me how to disabled this status line? I don't need it show overlap the printing screen.

[image]

guruathwal, 2020-01-27:
There is an option to disable in settings. Goto Setting>Feature settings

emtrax-ltd, 2020-01-27:
i think the visibility while "printstatus" is a bug.

billydragon, 2020-01-27:
Yes, don't need when printing because printing screen have all information. This display is helpfull when not printing only

bkzspam, 2020-01-27:
> There is an option to disable in settings. Goto Setting>Feature settings

I can't find that option. Where is it?

emtrax-ltd, 2020-01-27:
hide globalInfo while menuPrinting is active -> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/499e6f3533db9d9bde24d4c1dae92ef7b4268889

guruathwal, 2020-01-28:
@bkzspam  yes there is an option named: `Persistent Status Info` in `settings > Feature Settings` try clicking the page down button.

billydragon, 2020-01-28:
> hide globalInfo while menuPrinting is active -> [499e6f3](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/499e6f3533db9d9bde24d4c1dae92ef7b4268889)

Yes. Now it disappeared when printing and in main status screen. Thank you.

bkzspam, 2020-01-28:
> hide globalInfo while menuPrinting is active -> [499e6f3](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/499e6f3533db9d9bde24d4c1dae92ef7b4268889)

Thank you so much.

bkzspam, 2020-01-28:
> @bkzspam yes there is an option named: `Persistent Status Info` in `settings > Feature Settings` try clicking the page down button.

I suppose my firmware is different from you supposed. 
[image] (Features setting menu)

guruathwal, 2020-01-28:
@bkzspam The option is in List Mode. you are using the Icon Mode.

bkzspam, 2020-01-28:
> @bkzspam The option is in List Mode. you are using the Icon Mode.

You are correct. Then I can add it for the Icon mode.

mikefromups, 2020-02-25:
For version TFT Firmware version 3.0.25.1.

Remove Global Extruder and Bed temp from "Printing" Screen so it doesn't cover up your filename.  It is not needed on this screen since it is already in the center.

There is a new file in 3.0.25.1

TFT\src\User\Menu\Parametersetting.c

Change this line 
if(infoMenu.menu[infoMenu.cur] == menuMove || infoMenu.menu[infoMenu.cur] == menuStatus) return;

To This
if(infoMenu.menu[infoMenu.cur] == menuMove || infoMenu.menu[infoMenu.cur] == menuStatus || infoMenu.menu[infoMenu.cur] == menuPrinting) return;

closed 2020-08-14 by bigtreetech