bigtreetech/BIGTREETECH-TouchScreenFirmware #324: tft3.5-E3 V3.0 show only mode select menu.

GarageTech, 2020-02-01:
Just connect to Ender3 Creality 1.1.4 board.
Firmware update process made well.
Show only select mode menu.
After select Marlin mode - show on short time <1 sec Marlin screen and return to select mode menu.
Touch mode don't run at all.
Connect to EXP3.

Please help.

GarageTech, 2020-02-02:
Now unscrew from Ender case frame and make some touch on board, screen and re-insert display flat cable and finally image appear. But all these very unstable.

ntoenis, 2020-02-06:
Have a skr 1.4 turbo with the tft35-E3 V3 connected via (board)exp1-(screen)exp1 and (board)exp2-(screen)exp2. Screen wont turn on until I remove exp1 on screen and reput it in place.I can also just run exp3 with nothing else plugged in fine. Seems having exp3 populated and the touch screen black cable plugged in causes issues too. Can't seem to get both modes to work. Specifically marlin appearing blank and screen black

GarageTech, 2020-02-07:
I solve the problem when unscrew screen from standard mount. Possible mount always touch display module. Without mount all work fine. I think i need use some spaces between board and mount.

dziabi3, 2020-02-07:
I have the same problem (one of many).
In my case, removing the screen from the original ender 3 frame did not help.
Unplugging the tape from the exp3 socket helped. currently the display works only on TFT tape.

Interestingly, the tft35 e3 v3 display worked for 1 week without a problem.

ntoenis, 2020-02-10:
My problem appeared to be related to me having too many cables plugged in. All I need for both the stock cr10 marlin mode was exp1 to exp3. Then I ran the black tft cable with the reset pin removed. No exp2 cable connected and both sides are working perfectly.

dziabi3, 2020-02-11:
>>> Then I ran the black tft cable with the reset pin removed.

Why did you remove the reset pin?

ntoenis, 2020-02-11:
Based on the recommendation form a reddit user. Apparently it can cause the
board to freeze up and not load marlin like it should. Leaving it unplugged
seems to work so I’ve kept it that way

dziabi3, 2020-02-11:
Thank you for the information, I'll try too.

closed 2020-08-14 by bigtreetech