bigtreetech/BIGTREETECH-TouchScreenFirmware #325: [BUG] No access to TFT SD-Card  on TFT24 and SKR PRO 1.1 in touchmode

arminth, 2020-02-02:
### Description
In Touchmode, I can't get access to the TFT SD-card unless I unplug the EXP2 cable. 
Running latest Marlin 2.0.x bugfix. 

### Steps to reproduce

1. Try to acces TFT SD - read aerror.
2. Unplug EXP2 - it works

**Expected behavior**
No unplugging required

**Actual behavior**
unplugging EXP2 required

### Hardware Variant
TFT24
### TFT Firmware Version & Main Board Firmware details
TFT latest download
Marlin 2.0.x-bugfix download as of 2.2.2020
### Additional Information
Looks like related to #249

rndsum, 2020-02-03:
You have 
`//#define REPRAP_DISCOUNT_SMART_CONTROLLER `
Not disabled?  Other displays like the #define CR10_STOCKDISPLAY will not use the second or map the SD card pins. 

You might need a few other things configured for it I don't have that setup on mine as I am using a computer or pi to talk to it via usb.

arminth, 2020-02-04:
It worked exactly like this  on the SKR 1.3. I have Reprap discount full graphics controller enabled as it should be.

rndsum, 2020-02-08:
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/35 Might try that. 

Though checked mine and I was able to print using the marlin version of the SKR pro firmware on the SKR and this firmware on the LCD with all 3 cables connected.

zanvettor, 2020-02-15:
I have a similar issue. I can't print through LCD1286 mode, only with touch mode. Wich bothers me because touch mode isn't working with M600 (filament change)

Scope666, 2020-02-15:
I would try putting Marlin in CR10 mode.  #define CR10_STOCKDISPLAY and other display types commented out.  It should then work with one EXP cable in Marlin Mode.  (both modes working for me setup like this)

arminth, 2020-02-15:
It is not a CR-10 display. It has EXP1 and EXP2 and no EXP3. This can not (and does not) work.

Scope666, 2020-02-15:
I don't know about the TFT24 but with the TFT35 E3 it definitely works in CR10 mode with just the one grey cable.  (in Marlin Mode)

arminth, 2020-02-15:
I tried it and got nothing on the screen. No matter which cable I unplugged. Back to reprap full graphics and it worked.

Scope666, 2020-02-15:
This might explain it, looks like the E3 can do the one cable thing, I'm guessing the SKR 1.3 can't:

arminth, 2020-02-15:
Different story. My point was with skr pro 1.1. With skr 1.3 everything works flawless.

arminth, 2020-03-31:
Any news on this issue? EXP2-cable on SKR 1.1 pro competely blocks all access to SDCard on TFT. This is not acceptable!
@bigtreetech

oldman4U, 2020-10-09:
Hi. Has this issue ever been solved?

arminth, 2020-10-21:
Nope. still persists and no reaction from BTT.

oldman4U, 2020-10-21:
All I can offer you is a TFT28 or TFT35 E3 v3 and the SKR Pro v1.1 and to check if this works. But do you mean you can not see the TFT SD card under Print unless you do the trick, or is the TFT SD card there, but there are no files shown, or are there even files shown but you can not select them.

And of course you swapped the EXP cable(s)?

I am using Marlin 2.0.7.2 and the TFT firmware from yesterday.

Maybe I mentioned it before, but for me this sounds more like a hardware issue. Has this same TFT24 and cables been working with the SKR?

oldman4U, 2020-10-21:
I know you are waiting for BTT, but this is a misunderstanding. In case it is a software issue, the Open Source community has to fix it, in case it is a hardware issue, you have to contact the company you bought the hardware from.

Tested the TFT35 and it works flawlessly.

Tested TFT28 and works also great.

Have you checked if the problem also occurs in case the RS23 cable is not connected - just for a test.

And do you have something else connected to the TFT or Pro which could interfere? I hade a broken end stop cable which cause some funny issues...

arminth, 2020-10-21:
Just to make it clear: This is the same Display, which I used on the SKR1.3 no problem. Then I switched to something mor powerful AKA SKR 1.1Pro, but kept the rest (including the display) the same. On the SKR 1.1 Pro, it is definetly the case, that the EXP2-Cable/Port connected to the display prevents it from accessing the SD card in the display. When switching from 12864 mode (which works perfect) to touch mode with ans SD inserted, I get the message "SD Card inserted". But when you try to read from the card, you get readerror.
Same applies if I want to perform a firmware update.. I need to pull the exp2-cable, then perform the firmwareupdate, the replug.
It's not critical for me as I am printing thru Octoprint anyways but it still is strange.

closed 2020-12-27 by stale[bot]