bigtreetech/BIGTREETECH-TouchScreenFirmware #326: Option for center button for manual leveling

Phydeaux24601, 2020-02-03:
The four corner leveling option is great, but would it be possible to add an optional fifth button to check the bed center? (Marlin has it.)

MrTimPovey, 2020-07-19:
I have completed this myself for the TFT35-E3-V3.0 so it was more useful for my delta.

added by changing in TFT>src>User>menu>leveling.c to have a point_5 entry, and an additional item in the array, holding 0,0 and then an additional case statement.

edited language.inc to have the additional word

edited language_en.h to have the description, "centre"

added the same to icon_list.inc

added the bmp file to the folder.
[centre point level.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/4944732/centre.point.level.zip)

bigtreetech, 2020-08-17:
has merged in master branch， so close here

closed 2020-08-17 by bigtreetech