bigtreetech/BIGTREETECH-TouchScreenFirmware #328: tft24_V1.1.25.1 USB corrupt

lucsound, 2020-02-03:
When I print from usb it only heats the bed. and I stop printing shows loading ...
  In another gcode the printer starts to move randomly, and the display shows a flashing "nozzle cold stop" sign, but it keeps moving.

lucsound, 2020-02-04:
When I insert a 1GB micro SD card with an SD adapter, I can print the G-Code file perfectly without problems. But when I insert the micro SD card into a micro sd to USB adapter and place it in the USB port of the screen at the beginning of printing it does not raise temperature it starts to move in a random way and puts a message on the screen.

I have carried out tests with different pendrive of 1GB capacity; 2GB and 4GB.
With which I decided to try the same micro SD card in both inputs

Also note that it does not respect the career limits of the printer.

This has started from one moment to another has not coincided with any software update please someone can help me thank you very much.
[image]

ReubenBTalbott, 2020-03-07:
I have the same issue except it's even worse for me, nothing heats up and it does also does not home, it just starts printing.

oldman4U, 2020-09-05:
Hi.

What is the status of the reported issue? Have you tried with the latest firmware?

closed 2020-12-15 by stale[bot]