bigtreetech/BIGTREETECH-TouchScreenFirmware #333: [FR] Invert X axis controls

appideasDOTcom, 2020-02-10:
Through Configuration.h, you can currently invert the controls for the Y and Z axes, but there is no way to invert the controls for the X axis. Why?

**Describe the solution you'd like**
There needs to be a way to invert the controls for the X axis. Printers that home on the right side will have X axis movement from right to left, so the "+" pointing to the right and "-" pointing to the left is incorrect. That's a whole lot of 3D printers.

**Describe alternatives you've considered**
Unless I'm not seeing it, there is no option to invert the controls of the X axis, which is necessary for the arrows to point in the correct direction and have the correct labels for a very large number of printers.

appideasDOTcom, 2020-02-10:
I was able to get the exact behavior I (and others) need with some small changes to src/User/menu/Move.c

You can see the changes here:
https://github.com/appideasDOTcom/BIGTREETECH-TouchScreenFirmware/commit/ac14763e618549ebc746e883c7c05e893ef2ec14#diff-706125ec1ecde3b23cc2b967df715737

The changes that I made should happen through the use of an INVERT_XAXIS constant defined in Configuration.h rather than requiring a code change to Move.c

appideasDOTcom, 2020-02-10:
OK, so... If you want to add this to configuration.h:
`#define INVERT_XAXIS `

Then the changes in this commit will behave exactly as the INVERT_YAXIS and INVERT_ZAXIS features:
https://github.com/appideasDOTcom/BIGTREETECH-TouchScreenFirmware/commit/665b05c814fe66a9f6006c75ab998335317595a4

I've tested it both ways - making sure the original behavior is preserved without the flag, and that the desired behavior is achieved with the flag.

Should this be a pull request? It would be a pretty easy change to make manually.

guruathwal, 2020-02-11:
It is confusing.. will the printer with -X towards right print the models mirrored or not. I was unable to find printer with -X towards right. Although some printer home the X axis towards X Max but still the X axis direction remains same. The slicing software also takes the bottom left corner of the bed as (0,0).
[image]

The main motive for inverting the Y axis was to match the user perspective when the printers shape is different(like Ender or CoreXY). The actual Y axis direction remains same same goes for the Z axis where the perspective is based on movement of the Hotend or the Bed.

AnHardt, 2020-02-11:
The reasons lie a bit deeper.
All CADs/cams/slicers/routers/3D-printers assume to have a right handed coordinate system.
That implies, when x direction is the other way around and y direction stays the same, also z changes its direction.
Breaking that paradigm would be very confusing - and produce mirrored around one axis parts..

appideasDOTcom, 2020-02-11:
Why is it confusing to make the X axis work the same way the Y and Z axes work? The first Google search for "reset 3d printer origin" gave me this link:
https://drmrehorst.blogspot.com/2017/08/setting-up-3d-printers-origin.html
Which had this picture:
[image]
The second image is precisely what I've run on many printers for quite a few years.

If the very first Google result has a drawing of what I'm saying, I'm not making it up.

Here's one specific example from the commercial-space, if you really need to see one:
https://www.amazon.com/REPRAPGURU-RepRap-Printer-Plastic-Company/dp/B01KCYKNZ0/ref=sr_1_1?crid=3EHGVIG64PRQV&keywords=reprap+prusa+i3&qid=1581439480&sprefix=reprap%2Caps%2C236&sr=8-1

Why do you need to be concerned about where the slicer thinks the origin is in order to make this control consistent with the other controls? The origin and which motor direction is "left" and which is "right" are settings in Marlin, and have nothing to do with the slicer. My slicer "thinks" the origin is front left. It always has. The slicer is 100% irrelevant.

Many Prusa clones have inverted both X and Y axes - primarily because the printhead is in your way about 90% less often when you need to work on something that's sitting on the build plate - but the reason for moving the origin there is irrelevant.

@AnHardt No such assumption is made by "all" 3d printers or slicers. In fact, there's a whole class of printers that have the origin in the center of the bed. I'm not breaking any paradigm that wasn't already "broken" intentionally by Marlin - as a feature.

I really don't get what is so difficult about this. You have provided an easy way to invert the controls for two of three axes. Why is it difficult, confusing, or even a question  to make the third axis behave exactly the same? If, for nothing else, do it for consistency.

I've already given you the code. If other people don't want it, they don't have to enable the flag.

appideasDOTcom, 2020-02-11:
I'm sorry, but both of these comments make me believe that you don't understand how a 3D printer's coordinate system works, what role the firmware and slicer play, or why those things are completely irrelevant to this request.

Marlin was specifically designed so that the origin can be anywhere you like without changing your modeling or slicing software because:
1. The "+" and "-" (or MIN and MAX) directions can be inverted on any axis, and
2. The home direction can be inverted on any axis

grbl does exactly the same, and I have a CNC router that homes to the back left and another that homes to the front left. My slicer neither knows nor cares which direction is "home" or which direction indicates "+" or "-" on any axis because the firmware handles that.

I have personally run 3D printers with the origin at each of the 4 corners, and I've never had to tell my slicer that anything was different.

None of that has anything at all to do with which direction an arrow is pointing on a screen or what is on that button's label. All I'm asking for here is to swap the labels and function on two buttons on a touchscreen, and that has zero to do with how my printer is configured.

I'm really at a loss as to why implementing this is even a question. It's a very simple change that makes something that is currently inconsistent to be consistent.

AnHardt, 2020-02-12:
> No such assumption is made by "all" 3d printers or slicers.

It is! But in Marlin you can adjust the direction **to get a right handed system** - or to violate the rule by purpose.
Your second picture also shows a [right handed coordinate system](https://en.wikipedia.org/wiki/Right-hand_rule). Just seen - like from standing behind the printer. If the origin is in the center of the bed, or somewhere else, does not change the handedness of the system.

guruathwal, 2020-02-12:
@appideasDOTcom 
Your topic looked ambiguous as you stated in your opening comment : 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/333#issue-562262679
>There needs to be a way to invert the controls for the X axis. Printers that home on the right side will have X axis movement from right to left, so the "+" pointing to the right and "-" pointing to the left is incorrect. That's a whole lot of 3D printers.

There was no reference for the direction of Y axis, which led to the assumption that it is same as default directions.
It was in your later comment you stated to search for `reset 3d printer origin` for which it will be mandatory to invert the Y axis controls also (please correct me here if I am wrong), due to which this has to be more than just feature to invert X axis controls only. 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/333#issue-562262679
>The more logical change seems to be to add some defines between lines 13 and 35, where the other INVERT axis ifs already exist, but I'm not sure why this feature should be exclusive of MENU_LIST_MODE

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/333#issuecomment-584733866
>I really don't get what is so difficult about this. You have provided an easy way to invert the controls for two of three axes. Why is it difficult, confusing, or even a question to make the third axis behave exactly the same? If, for nothing else, do it for consistency.

The feature is not added till now because none of the users have requested it till now and adding more code just for consistency when no one has felt need for it till now on devices like these TFTs with a very limited programming memory will make the memory run out soon. The Flash storage is already filling up with too many icons.
This is also the reason I added the list view to remove the need of icons for every single option. The List Mode in Feature settings has been set as permanent in PR #302 .
But still if you can make the icons and add code for these setting in icon mode then I will definitely appreciate it. 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/333#issuecomment-584744424
>I'm sorry, but both of these comments make me believe that you don't understand how a 3D printer's coordinate system works, what role the firmware and slicer play, or why those things are completely irrelevant to this request. 

what do you want to imply with this?

appideasDOTcom, 2020-02-12:
> It is! But in Marlin you can adjust the direction **to get a right handed system** - or to violate the rule by purpose.(*)

Your latter statement proves that your first statement is incorrect. Consider this:
```
if( bobWants( "peach pie" ) ) { give Bob peach pie } 
else if( bobWants( "pecan pie" ) ) { give Bob pecan pie }
```
The fact that there is a choice means that the developer is making NO assumption about Bob's preference for pie, despite the fact that he may hate pecan pie, and may think that everyone who likes it is wrong.

The fact that Marlin gives you the ability to flip every axis and homing direction PROVES that they made absolutely zero assumptions about any user's chosen coordinate system. In fact, the way they implemented that very FEATURE/CAPABILITY is by giving configurable options that decouple the machine configuration from the slicer. If the firmware is correct, the slicer doesn't need to know anything about the printer's implemented coordinate system. It's that way by design.

However, because Marlin's implementation of the coordinate system is entirely independent of the slicer and the direction buttons on the touchscreen (a point I made in initial my follow-up comment), continuing this discussion has nothing to do with my request, and i would like to end it, please.

P.S. If you want to see an example of how someone configured Marlin for a left-handed coordinate system without changing his slicer, here is something:
https://3dprinting.stackexchange.com/questions/5878/change-slic3r-settings-for-left-handed-coordinate-system
I've done it in the past for testing, but it wasn't better than RHS with a back-right origin. I still have a CNC router that runs a LHS, and its toolpath generator is completely oblivious to the fact. Again, this is not important to the issue, so we should close this part of the discussion.

appideasDOTcom, 2020-02-12:
> There was no reference for the direction of Y axis, which led to the assumption that it is same as default directions.
> It was in your later comment you stated to search for `reset 3d printer origin` [...]

That is incorrect. Please remember that we are talking about this **only**:
Switching the label and "sent gcode" of two buttons on a touchscreen

The ability to flip one axis should be independent of any other axis. If a user wants/needs to flip X and not Y, why not allow it? The function of those buttons on the screen has ABSOLUTELY NOTHING to do with whether or not the machine is printing in the correct direction. Why are those issues getting confused with each other?

When I push the button labeled "X+" (with the distance set to 1mm) this gcode is sent to the printer:
G1 X1
Why does the touch screen care what happens at that point? Does it care that the icon is a left pointing arrow or a right point arrow? No!
If I need the button on the left to be labeled "X+" and to send the command `G1 X1` and I need the button on the right labeled "X-" and to send the command `G1 X-1` what's the problem? How is reassigning the label and resulting gcode going to affect anything related to the way the printer prints?

It's isn't!

> The feature is not added till now because none of the users have requested it till now and adding more code just for consistency when no one has felt need for it till now on devices like these TFTs [...]

I will look into it as soon as I get a chance.

> 
> > I'm sorry, but both of these comments make me believe that you don't understand how a 3D printer's coordinate system works, what role the firmware and slicer play, or why those things are completely irrelevant to this request.
> 
> what do you want to imply with this?

There was no implication. It was a definitive statement. The function of these buttons has absolutely nothing to do with the configured coordinate system of the printer or the printer's ability to print correctly.  Also, it's a really, really small and simple change with zero negative impact. I wasn't expecting to have to spend time talking about/defending coordinate systems when they have nothing to do with my request. The fact that the topic of printer coordinate systems got drug into my unrelated issue is evidence of the truth of my statement.

I'm not trying to pick apart anyone's good work, and I certainly appreciate what BTT has done and is doing. I just want a solution to this very minor request without having to spend a lot of time talking about unrelated topics.

Thank you.

guruathwal, 2020-02-14:
@appideasDOTcom  No one has asked you to stop it or not to allow it.. when you post in a public forum area you should expect both types of comments that are in favor and those that are opposed.

appideasDOTcom, 2020-02-14:
> @appideasDOTcom No one has asked you to stop it or not to allow it.. when you post in a public forum area you should expect both types of comments that are in favor and those that are opposed.

Yes, but the opposition was to a topic that's entirely unrelated to my request. It was a simple request that has been confused by 100% irrelevant debate.

appideasDOTcom, 2020-02-14:
This pull request should do the trick:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/338

guruathwal, 2020-02-16:
@appideasDOTcom You did not updated this line in your PR #338 
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/5777f41b5f8c1a41410a6874614499c91ac78fa2/TFT/src/User/API/flashStore.c#L6

When adding or removing a setting storage parameter the above line also need to be changed to prevent the firmware from loading settings from wrong address. The number is in date in format of `YYYYMMDD`

appideasDOTcom, 2020-02-18:
I missed that (obviously). Thanks for pointing it out. Did you take care of it, or should I update the branch that has the pull request?

guruathwal, 2020-02-18:
No I did not updated it.

appideasDOTcom, 2020-02-19:
That change was committed to the PR branch. Thanks!

closed 2020-08-17 by bigtreetech