bigtreetech/BIGTREETECH-TouchScreenFirmware #335: [Q] No warning when thermal runaway protection active

benemos, 2020-02-10:
BTT TFT35 V2 - Firmware BIQU_TFT35_APP1_V2.0.25.1.bin

I tried to test the thermal runaway protection today on my CE3 Pro with a BTT SKr Mini E3 board. I did this by removing the thermistor from the hotend block and requesting a hotend temp of 180. Waited 5 minutes and no warning on the BTT TFT35. I tested the temp of the block with a temperature probe and it was clear the block had stopped heating after getting to around +60C (thermal runaway protection had activated) but there was no alert/message of this on the TFT display. I've looked at the software but can't find the correct setting I'd need to check or enable. Any guidance or is this a known issue? Also is this an issue within my Marlin build or the TFT config files? I can attach files if someone can tell me exactly what I need to attach?

Thanks.

Hans007a, 2020-02-11:
check it with the pronterface and post here the messages, that marlin is sending to this

benemos, 2020-02-12:
Thanks,

Pronterface returned the following;

Connecting...
Printer is now online.
Setting hotend temperature to 180.000000 degrees Celsius.
Error:Heating failed, system stopped! Heater_ID: 0
[ERROR] Error:Heating failed, system stopped! Heater_ID: 0

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

Hans007a, 2020-02-15:
In the file parseACK.c in the line 264 (at my version of the file) is this:
```
else if(ack_seen(errormagic))
    {
      ackPopupInfo(errormagic);
    }
```
in parseACK.h you can find the definition:
`static const char errormagic[]        = "Error:";`

The TFT firmware should interpret any error, but I think, this is a mistake by the marlin because no space between "Error:" and the message...

closed 2020-08-17 by bigtreetech