bigtreetech/BIGTREETECH-TouchScreenFirmware #336: TFT Print screen not working after selecting OnboardSD in print menu

sides580, 2020-02-10:
#245 # Description
Watch video from link at bottom to see the problem and avoid reading. 
After touching the "OnboardSD" print button I get the correctly loaded .gcode files from in my SKR pro 1.1 board sd port. However, the screen sorta locks up. It's like a hidden menu is in the center of the screen. The back button doesn't work. If i touch randomly around the screen sometimes I get a popup menu which I am no able to clear without resetting the screen. If If switch to marlin screen mode, I am able to load from SD card and start a print without any problems. I am also able to print from the TFT SDcard on the TFT35. I've already had a few lockups with the TFT35 and want to print from the SKR sdcard for reliability reasons. 

### Steps to reproduce

1. Update firmware to latest from github. "Copy to SD Card root directory to update - Unified Menu Material theme"
2. Select correct baud rate. Tried both 250000 and 115000
3. Tried some other bug fixes listed on this github. Will update this with those details.

**Expected behavior**
I expect to be able to select a file.
I expect the back button to work. 
I expect the scroll buttons to work.

**Actual behavior**
Unable to pick a file and start a print.
Scroll button doesn't work if enough files are on the card.
If touch middle of the screen, I get a strange popup window. Unable to click the ok button to clear the popup window.

### Hardware Variant
TFT35 3.0
Dsoftware version =25.1

### Additional Information
Here is a video of the problem. Clear as day. You can briefly see the popup screen i'm talking about.
https://drive.google.com/open?id=1G6gJ9BpZ9Ezl7YGCtVGpnpjg_9rn_B83

Picture showing screen loads. F
[image]

Strange popup when clicking in middle of screen:
[image]

sides580, 2020-02-10:
I have a minor theory. I'm using marlin 2.0 as a CNC router not a 3d printer. I'm wondering if something from inside the main loop is interfering because of an unusual setting.

dziabi3, 2020-02-11:
I have a slightly different problem (one of many - TFT35 E3 V3.0).
When I am looking to print from "OnboardSD gcod has been loaded but does not start.
The table temperature and hotend are 0.

sides580, 2020-02-11:
> I have a slightly different problem (one of many - TFT35 E3 V3.0).
> When I am looking to print from "OnboardSD gcod has been loaded but does not start.
> The table temperature and hotend are 0.
This sounds almost the same issue as me. I suspect this is a real problem and most people ignore it by printing from the screens SD card.

sides580, 2020-02-16:
SOLVED, but solution not implemented in current firmware. Band aid solution below!

Ok, I found the problem this is really annoying situation. I don't know how to connect a debugger cable to these things so I did a bruit force approach to finding the line of code that broke my TFT35 screen when trying to print a program from the onboard SD card on my SKR Pro 1.1.

Commit this line of code out and all problems go away.
Look inside "ParseACK.c" Its inside the Menu folder
Commit line: 278. Might be a different line for you.
ackPopupInfo(echomagic);  <--- This fucking line of code
Change to //ackPopupInfo(echomagic);
Note; comment the one inside the: "else if(ack_seen(echomagic))"

The problem appears to be that my controller is sending an echo command back with a warning from M20 command sent on bootup. Then when going into the SD card print screen this popup message is somehow taking over the screen but the popup doesn't get displayed correctly. Then if you manage to clear the popup it comes back instantly. It has to do with how the software sends a M21 command and waits for a response. I didn't dig any deeper, just happy I found a Band aid!!

Dev's need to handle something differently here. Personally, this echo popup is just annoying to me and happy I committed it out.

dziabi3, 2020-02-20:
I decided to test TFT35 E3 V3.0.
At the beginning I disconnected the reset signal and the display began to work stably (no longer stuck on the mode selection screen)

Encouraged by this fact, I followed your instructions and unfortunately it did not help. Gcod is started, but the set temperatures are 0, the actual temperatures are not refreshed and the fan speed is 0.

I need to do more tests but I am increasingly convinced that I can't trust the TFT35 E3 V3.0 display.

For the past two weeks I have had Marlin mode and the original Ender 3 display. I didn't have a problem.

Thank you for your help.

oldman4U, 2020-10-09:
Hi. ACK is now handled correctly. Could you please check and close this ticket afterwards.

Thank you

closed 2020-12-15 by stale[bot]