bigtreetech/BIGTREETECH-TouchScreenFirmware #337: BUG - TFT35-E3 V3.0 only displays LCD Simulator message in Marlin mode

Andy-ABTec, 2020-02-12:
Hi,
Just purchased a BTT TTF35-E3 V3.0 to use on my Ender 3 along with an existing BTT SKR 1.3 board.

Touch Mode appears to be fine, but Marlin Mode just presents the Simulator message, the rest of the screen is blank.

All firmware is the latest available, Marlin 2.03 for the SKR and BIGTREE_TFT35_V3.0.25.1 for the TFT. I'm using the Unified Menu Material theme, and the boards are interconnected with all three cables (correctly orientated on EXP1, EXP2 and RS232/TFT).

No amount of encoder twiddling or clicking has any effect other than holding down for 3 seconds and switching back to Touch Screen mode.

Thoughts?

hfog, 2020-02-12:
The Ender 3 only uses 2 cables. The screen cable (you can use the same one that you had connected to your old display) plugs into EXP3 and the RS232/TFT cable connects between the RS232 port on the TFT and the TFT port on the SKR. The EXP cable is what is used to drive the Marlin mode, in which case the TFT is just simulating the dumb terminal behavior of the old screen. The RS232 cable is what the touch mode uses, in which case the TFT acts as a connected computer, similar to Octoprint.

Andy-ABTec, 2020-02-13:
Many thanks for the reply, however I'm using an Ender 3 chassis fitted with an **SKR 1.3 board**, and everything I've read/watched indicates I should connect EXP1, EXP2 and RS232/TFT connections between the SKR and the TFT.

hfog, 2020-02-13:
You should give it a try. I think you'll find that you're mistaken. I'm also using an SKR board.

Andy-ABTec, 2020-02-13:
OK, but now I'm confused, which connector on the SKR V1.3 (EXP1, EXP2, or AUX1) should I connect to EXP3 on the TFT?

hfog, 2020-02-13:
Is it an SKR Mini E3 board? That's what I assumed since you're dropping it into an Ender. That connects from EXP1 on the SKR to EXP3 on the TFT. If it's a different SKR model then you may be correct. Sorry if I've created any confusion. Hopefully you get it figured out.

Andy-ABTec, 2020-02-13:
Its a standard SKR V1.3 board, I'm 99.9% sure it's correctly connected, EXP1 to EXP1, EXP2 to EXP2 and TFT to RS232. I just can't get any functionality in Marlin Mode, just a blank screen except for the "LCD12864 Simulator" heading. :confused:

hfog, 2020-02-13:
Yeah, if the touch screen mode is working but the Marlin mode isn't then the TFT to RS232 sensor is the piece that isn't working. I should have read your initial post more closely.I thought your problem was with the touch screen. I assume you've tried reversing the TFT connection. Right?
Anyway, sorry for muddying the waters.

Andy-ABTec, 2020-02-13:
Tried that, just get a continuous beep...

KennyD42, 2020-02-14:
I have this very same problem. All documentation and all video guides I've watched say the black cable plugs into the RS233 on the TFT35 E3 board and into the TFT connection on the SKR 1.3. It also says you are supposed to connect both EXP1&2 into the same ports on the SKR 1.3 respectively. It's described as "plug & play" for the Ender3. I can't get any definitive fixes on this either.

Andy-ABTec, 2020-02-14:
New definition of "Plug & Play" - plug it in and play for days trying to make it work... 😠
I'm seriously considering returning my TFT to Amazon for a refund, unfortunately my SKR V1.3 is past its return date...
I'm pretty sure the issue is with the SKR's EXP1 and EXP2 sockets. Comparing the pin diagrams for the TFT and the SKR there is definitely something screwy going on,
BigTree's Tech Support seems to be non existent and they appear to be relying on customers to do their QA and fault finding.
I've also had a look at the "documentation" (I use the term loosely) for the SKR V1.4 as I was considering trying one as it appears to be an "updated" version of the 1.3, but the EXP pin-outs and interconnections with the TFT still seem screwy and until a circuit diagram is released I'm not wasting my money.
Seriously BigTree, I'd expect the various bits of BigTree hardware to play nicely together, or if they don't at least to admit there's a problem and document a fix or work-around not just ignore it and keep taking peoples money

Scope666, 2020-02-14:
If you have an Ender 3 stock screen, for Marlin mode it should be only one EXP cable, not 2.  In Marlin this is configured as  #define CR10_STOCKDISPLAY

https://i.redd.it/pd14hsxfvl631.jpg

Andy-ABTec, 2020-02-14:
OK, so forget the Ender 3 for a minute...
I have a SKR V1.3 and a TFT35-E3 V3.0 on the bench interconnected EXP1 to EXP1, EXP2 to EXP2 and RS232 to TFT.
In Marlin I have made the following changes
```
#define SERIAL_PORT -1
#define SERIAL_PORT_2 0
#define BAUDRATE 115200
#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_3
#define MACHINE_UUID "31963356-fd60-4aa1-b0c8-deee061b0c39"
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
#define BLTOUCH
#define NUM_SERVOS 1
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
```
And on the TFT35
```
#define ST7920_BANNER_TEXT "LCD12864 Simulator"
```
Everything seems to function normally in Touch Mode but in Marlin Mode all I see is "LCD12864 Simulator" and a blank screen.
**NOTHING** I have watched or read has been helpful so far, including rotating the EXP connectors on the SKR 1.3.

bumpaneer, 2020-02-14:
Dumb question, but it got me when I first switched displays. Did you rotate the knob when switching to Marlin Mode? By default its a black screen for me until I rotate the knob.

Andy-ABTec, 2020-02-14:
#define ST7920_BANNER_TEXT "LCD12864 Simulator" in the TFT35 config proved that the display was "live"-ish but no ammount of knob twiddling or pushing helps...

hfog, 2020-02-14:
One thing occurs to me. You mentioned that your SKR board is newish, right? It's possible that it isn't a problem with the TFT and that your Marlin firmware on the printer board isn't configured properly to work with this display.

Andy-ABTec, 2020-02-14:
No amount of knob twiddling or klicking helps...

KennyD42, 2020-02-17:
So I'm guessing BigTreeTech still has no solution for this problem. I also cannot return my SKR1.3 if the problem actually is in that piece of hardware. Please help us. It's obvious this is a common problem and all we want is a way to use your hardware, the way you advertise it works. That's why we gave you our money to get it.

wallen83, 2020-02-17:
> OK, so forget the Ender 3 for a minute...
> I have a SKR V1.3 and a TFT35-E3 V3.0 on the bench interconnected EXP1 to EXP1, EXP2 to EXP2 and RS232 to TFT.
> In Marlin I have made the following changes [...]

have you put // before #define CR10_STOCKDISPLAY in marlin?   I think you have to disable it when you have enabled REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Andy-ABTec, 2020-02-17:
Yes it is commented out. I'm pretty sure that all the displays are commented out in the default Configuration.h.

But I **Fixed it!** See [HERE](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/issues/9#issuecomment-586908632)

closed 2020-02-24 by Andy-ABTec

bryanhunwardsen, 2021-02-14:
I just want to add an alternate solution for anyone facing this issue where the listed solution was not the issue.

When switching to emulation mode the screen does not load:
A simple issue I faced and resolved, the marling cfg.h did not match the TFT cfg.ini for the (simple) screen type.
Marlin was set to LCD2004 TFT was LCD12864, flashing either/both to unify to the same screen type corrected the issue.
In my case I re-flashed the TFT to LCD2004 as no recompile required and LCD2004 is a better utilization of the TFT35 screen realestate.