bigtreetech/BIGTREETECH-TouchScreenFirmware #342: [BUG] "Failed to enable Bed Leveling" TFT35 V3.0.25.1

tmarcus68, 2020-02-19:
Upon completion of flashing "BIGTREE_TFT35_V3.0.25.1.bin" on TFT35 V3.0, error prompted on restart "Failed to enable Bed Leveling echo:Bed Leveling OFF echo:Fade Height OFF ok"

Using on SKR V1.3 (marlin 2.0) with TFT35 (V3.0.25.1)

"BIGTREE_TFT35_V3.0.25.1.bin" Firmware is pulled from github. Tried self compiling by turning line 94 "#define AUTO_BED_LEVELING" off, will not received the error prompt. 

But what is the implication of turning "#define AUTO_BED_LEVELING" off? Based on my understanding the printer will not send command M500 to store the leveling values.

Any one else faces this issue?

Scope666, 2020-02-20:
If you do have a probe, you'll get this error if you haven't probed the bed and done an M500 to store the results afterwards.  In other words it's saying you don't have a stored mesh.  Once you go a G29 / M500 the error should go away.

guruathwal, 2020-02-20:
This is not a bug. As @Scope666 has pointed out, if you do not need to store mesh then Just below the line 94 there is line number 98 which you need to comment out to stop sending `M420` on startup:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/5777f41b5f8c1a41410a6874614499c91ac78fa2/TFT/src/User/Configuration.h#L96-L98

thisiskeithb, 2020-02-20:
This really should be patched so it doesn’t error first thing on boot.

tmarcus68, 2020-02-22:
@guruathwal so its still a bug because if i need to store the mesh which base on my understanding is needed for ABL i cant comment out line 98. correct me if i am wrong, still very new to 3d printing.

guruathwal, 2020-02-22:
@tmarcus68 even then its not a bug. The error is generated by Marlin firmware.
If you have enabled mesh saving on marlin, then on receiving M420 command, the marlin firmware will check for stored mesh data. If it does not finds a valid mesh then it will send the said error message to the host.

tmarcus68, 2020-02-22:
so its a marlin firmware issue not bug. ok note then i think can close the problem then. thanks all for the help.

closed 2020-02-22 by tmarcus68

Juan1Carlos, 2020-10-24:
Hola tengo ese problema y Ya quedo claro.  (para poder imprimir, me las he visto algo mal)(desactivando G92 en repetier)

ahora el tema es el siguiente: cada vez que intento hacer un G92, ( X va en sentido contrario y no puedo nivelar la cama) intente invertir los motores después la lógica y nada, soy torpe para resolverlo.

Rhragnarok, 2021-05-02:
So wait, is this feature for like a BLTouch?  Or is this something for people who don't have a BLTouch (or the like)?   I am getting this error right now but I do not have a BLTouch or sensor.  I am just manually leveling the bed atm.  How do I get rid of the message and still have the bed leveling wizard?

jmger, 2023-03-06:
This is from the marlin page under M420. 
I am running Marlin 2.1.x bugfix -date: about feb-27-2023

"[S<bool>]	
Set enabled or disabled. A valid mesh is required to enable bed leveling. If the mesh is invalid / incomplete leveling will not be enabled." 
So this also appears that the error is that you need to set the probe height then run a Bedlevel to set/calibrate the printer. For the First time. Remember to save settings. I just turned on my printer to verify. And I am NOT getting the error message anymore.
TIP: for the EXABL Probe: if the bed leveling fails half way through the manual/auto Mesh/grid bedleveling process; check if the light on the probe is on or off. If it is off then you have to adjust the probe sensitivity so the light is on. Then try the process again. When the nozzle it close to the bed red LED light should be on. It would be very helpfull if marlin cout output this message... "Error: Auto bedleveling failed to initalize please run a mesh auto bed calibration" or some such message. Then we would not have to come here and spend countless hours in research and frustration.