bigtreetech/BIGTREETECH-TouchScreenFirmware #344: [BUG] Temps/Progress not updating in Touch Mode

pdeswardt, 2020-02-21:
### Description

In Touch mode, when directly printing a model, the bed and nozzle temps shows room temp until final temp is reaached and only then shows correct temp, sometimes it doesn't even show the final temps. Also - the progress of a print stays on 0% and the Z-position also stays on 0 and is not updated as the print progresses, None of these issues are evident in Marlin Mode.
[image]

### Steps to reproduce

1. Select Print form SD -> Temps only shows final temp when reached, and doesnt progress gradually as the bed/nozzle heats up. Sometimes it doesn't even update final temp at all, you will also see the progress % not updating and also Z-position ( layer height ) stays on 0

**Expected behavior**
When selecting any print, the target temps together with the current temp as is increases should be updated all the time as with Marin Mode. Also as print progresses, the % should increase and not stay on 0%. Also - Z-position should update as it progresses - it currently stays on 0% in Touch Mode.

**Actual behavior**
Target temp intermittently not updated in realtime - only when target temps reached.
Progress % is not updated and neither is Z-position ( layer height )

### Hardware Variant
BTT TFT 3.5 V 3.0
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0.25.1
BTT SKR Mini E3 1.2 running BTT Marlin 2.0.x bugfix with mesh leveling

### Additional Information

hfog, 2020-02-21:
This should be fixed in the next version. I suggest you pull guruathwal's master branch (https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware) and build that. It should fix your problem.

pdeswardt, 2020-02-21:
I did that , and my display is all gibberish....any idea why ? I'll try again in the meantime

Thanks

hfog, 2020-02-21:
Did you also flash the resources?  The icons have changed.

pdeswardt, 2020-02-21:
Hi, sorry mind explaining - im fairly new ....do i need to copy anything with the compiled bin file to the root of the SD card ?

pdeswardt, 2020-02-21:
Nervermind - i think u mean copying the TFT3.5 folder with the bin file.... ?

pdeswardt, 2020-02-21:
Works perfectly now..... thanks !!

hfog, 2020-02-21:
Yep. You figured it out.

pdeswardt, 2020-02-21:
It seems to have worked perfectly for the first couple of prints, however now it seems to have lost the temp readings, and progress %. Also intermittently hangs on "Loading..." trying ti read the SD card, o i have to power cycle it each time..... frustrating..

pdeswardt, 2020-02-21:
Ahhhh..i see whats happening here. It loses info if you print from the SKR board's SD card slot. If you print from the TFT SD card slot it works. However, trying to print from SD->USB thumbdrive fails too..

closed 2020-08-17 by bigtreetech