bigtreetech/BIGTREETECH-TouchScreenFirmware #345: [BUG] extruder temperature value overlays the icon

syampolsky, 2020-02-22:
### Description

Using 3.0.25.1. When heating extruder over 99 degrees, the third digit of the temperature is painted over the extruder temp icon.

### Steps to reproduce

1. Tap on T0 icon
2. Tap on Heat icon
3. Set temperature to 100 degrees(or any value >99)
4. Wait for the temperature to reach 100. The first digit paints over the right edge of the extruder icon in the upper right corner of the screen

**Expected behavior**
There needs to be enough space to allow 3 digit temperature indication without painting over the icon.
Changing VALUE_NOZZLE_X definition in Parametersetting.h 
from:
#define VALUE_NOZZLE_X  ICON_NOZZLE_X+BYTE_WIDTH
To:
#define VALUE_NOZZLE_X  ICON_NOZZLE_X+BYTE_WIDTH+BYTE_WIDTH/2
fixes the problem.

### Hardware Variant
BTT-TFT35-E3-V3.0
### TFT Firmware Version & Main Board Firmware details
3.0.25.1 on TFT
Marlin bugfix-2.0x branch on BIGTREETECH-SKR-mini-E3

pdeswardt, 2020-02-24:
Yeah - also noticed this

mikefromups, 2020-02-25:
For version TFT Firmware version 3.0.25.1.

Remove Global Extruder and Bed temp from "Printing" Screen so it doesn't cover up your filename.  It is not needed on this screen since it is already in the center.

There is a new file in 3.0.25.1

TFT\src\User\Menu\Parametersetting.c

Change this line 
if(infoMenu.menu[infoMenu.cur] == menuMove || infoMenu.menu[infoMenu.cur] == menuStatus) return;

To This
if(infoMenu.menu[infoMenu.cur] == menuMove || infoMenu.menu[infoMenu.cur] == menuStatus || infoMenu.menu[infoMenu.cur] == menuPrinting) return;

syampolsky, 2020-02-26:
> Remove Global Extruder and Bed temp from "Printing" Screen so it doesn't cover up your filename. It is not needed on this screen since it is already in the center.
good find!

closed 2020-08-17 by bigtreetech