bigtreetech/BIGTREETECH-TouchScreenFirmware #348: [BUG] (short description)

Skybeat, 2020-02-25:
### Description

I try to use the TFT35 V3.0 E3 on my Ender 3. I have a original Creality Board in version 1.1.5 with Marlin installed.
After i connected the TFT35 to my Ender 3, i can only use the Marlin mode. In grafical mode it shows all the time "No printer attached" also if i change the baud rate in the TFT settings.

Next is that i can to a firmware update because 6 different SD-cards show the message "read TFT sd card error" when i try to look into the SD card via grafical menu->"print"->"TFT SD".

If i shut down the printer, add a SD card with the firmware "BIGTREE_TFT35_V3.0.25.bin" and folder "TFT35" on it, into the TFT and start the printer again, the TFT shows a message "Card insertat" but never start a firmware update.

I was think that the TFT is broken and ordered one more but i have exactly the same issues.

- So I can't do a firmaware update
- SD cards will not be read
- no grafical menu with my original Ender 3 (Board v1.1.5 from Creality)

Skybeat, 2020-02-26:
update:

Now i know that the graphical mode dosen't work with the original board, so i have ordered a SKR mini E3 v.1.2.

But the firmware update is not posible with all my SD cards. Even if i change the partion on the card and create only a smal partition about 256 MB it wan't work.

A few minutes agao i received my SKR mini E3 v.1.2 and i used the SD card what comes with the board (it is a 128 MB Card.
With this card i was able to update the firmware of my TFT35 E3 v.3.0.
So it seems that the TFT35 E3 v.3.0 hase a problem with bigger and faster SD cards. Is there some limitations for SD card?

Thanks,
Nico

fenix8k, 2020-03-16:
Hi, I just create this account so no one struggle with this anymore.

In order any SD card can operate with this TFT Screen It need to be Less than 8gb Total size (acording to BTT), have a MRB partition table, be marked as active and primary, be formated as FAT. With this conditions I use a 1GB SD card to update the firmware (before all abobe, this card didn't work at all even in FAT format)

Hope this Help!

closed 2020-08-17 by bigtreetech