bigtreetech/BIGTREETECH-TouchScreenFirmware #354: [BUG] (short description)

2BWorks, 2020-02-29:
I have bought TFT 35 E3 version. The encoder knob doesnt work. It does not register when im rotating. What can i do to fix?

GeminiServer, 2020-02-29:
In simulation mode e.g. with marlin or?

2BWorks, 2020-02-29:
It doesn't work at all

fishwhisperer83, 2020-08-01:
Has this issue been resolved? I'm having the same issue, doesn't work in marlin simulation or changing between the 2 modes. This has happened with 2 boards, one I replaced because i thought it had a bad sensor. Using a TFT35_E3_V3.0 running TFT35_E3_V3.0.26.x Jun 15 2020

oldman4U, 2020-09-03:
@2BWorks 

What is the status of the reported issue.

Please close this ticket if you do not need it anymore

Thank you

oldman4U, 2020-09-05:
@2BWorks 

Close this ticket

oldman4U, 2020-09-11:
@bigtreetech 

Please close this ticket because a possible reason is described in the ReadMe and the user does not reply anymore.

Thank you

oldman4U, 2020-10-06:
Please help the community and close the ticket once you do not need it anymore. Thank you

npmomchev, 2020-11-20:
I have the same problem. Works only for choosing between simulation or touch. In Marlin simulation mode does not work

npmomchev, 2020-11-20:
**Solved**

Detached EXP2 cable.

closed 2021-01-26 by stale[bot]