bigtreetech/BIGTREETECH-TouchScreenFirmware #355: [Q] platformio.ini TFT35_V3.0 should use STM32F207VC

Andrei-Pozolotin, 2020-03-03:
@guruathwal 

Gurmeet:

1. `platformio.ini` says:
```
# BIGTREE_TFT35_V3.0
[env:BIGTREE_TFT35_V3_0]
board         = STM32F103VC
```

2. despite `TFT35_V3.0` actual hardware using:
```
board         = STM32F207VC
```

3. I am curious why not `STM32F207VC` ?

guruathwal, 2020-03-03:
I do not know the exact reason and also I do not have the V3 to notice the difference, but someone on Facebook earlier answered the same question with the reason that `STM32F207VC` is not listed in pre-configured boards in PlatformIO and both of these boards have the same core `Arm® 32-bit Cortex®-M3 CPU`. 
Also, the IO Pins for V3 in the firmware source are set up according to the `STM32F207VC` by BTT.

You can check out the Facebook discussion here:
https://www.facebook.com/groups/505736576548648/permalink/798781007244202/

and here is the link to the Boards page for Platform IO
https://docs.platformio.org/en/latest/boards/index.html#st-stm32

Andrei-Pozolotin, 2020-03-05:
1. thank you for the answer. perhaps this should be converted to "[FR]"
since at some point there will be need to switch to proper hardware chip config

2. on related topic: I have done some heavy hacking on my fork, 
the changes are too wild to qualify for normal PR, so I will post them here 
as [FR] issues instead, if that is ok.

AntoszHUN, 2020-03-31:
I'll change it to STM32F207VC. Then it doesn't translate correctly.

Andrei-Pozolotin, 2020-03-31:
> Then it doesn't translate correctly.

what do you mean?

AntoszHUN, 2020-03-31:
Does not translate. Unsuccessful. The build failed.

Andrei-Pozolotin, 2020-03-31:
Ah, you meant "does not compile".

AntoszHUN, 2020-03-31:
Write error to multiple files. If I give back the STM32F103VC then good. Interesting thing.

bigtreetech, 2020-08-17:
In fact, this project only uses the compilation chain of stm32cube framework, and does not use the `framework-stm32cube `library， it's use `Standard Peripheral Library` in [here](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/TFT/src/Libraries/fwlib). 
And the `framework-stm32cube` and `Standard Peripheral Library` of stm32f2xx have duplicate definitions (function name / variable name / macro define, etc.), so it will cause compilation errors.
so I set the mcu to STM32F1 not F2, because the `framework-stm32cube` of F1 is more clean

closed 2020-08-17 by bigtreetech