bigtreetech/BIGTREETECH-TouchScreenFirmware #357: [BUG] (short description)

nsmdigitalpc, 2020-03-05:
### Description
Hi, got a question. my board is a Bigtreetech gtr v1.0,    6 tmc2208 uart mode drivers, I have verified that everything works on the printer through pronterface.  My display is bigtreetech tft3.5 E3

### Steps to reproduce

1. download the "BIGTREETECH-TouchScreenFirmware-master"then made sure my cables are correct. tft connection & esp1
2. [updated the firmware with the classic icons & set the rate 115200
3. the touch screen works but does to control anything
4. the tft3.5 will not connect "no printer attached" to the GTR board. What am i doing wrong or missing please help

UnderXirox, 2020-03-05:
Hello do your have
edit your Marlin file Configuration.h file
#define SERIAL_PORT 6 // for Wifi (1 for TFT uart)

#define SERIAL_PORT_2 1 // -1 for USB 6 for esp on the motherboard
// I connect the 3 cables EXP1 EXP2 and the art in this way I can switch between the mode

//  6 = esp-1s
// -1 = octoprint on a Raspberry Pi 3 (exemple SKR-PRO)
//  1 = TFT

// define SERIAL_PORT 6 // 6 = esp, -1 usb on my skr it's up to you to find the serial port for usb, 1 = TFT

sorry m'y English

tbojim1975, 2020-03-05:
Marlin Configuration.h in the bigtreetech gtr or marlin in the bigtreetech tft3.5 v3.0?

tbojim1975, 2020-03-05:
Never mind it's been a long day :-) I figured it out . I will let you know the outcome. Thank you for this information

UnderXirox, 2020-03-05:
Dans Marlin Configuration.h

`/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 6  // pour le Wifi car Carte Buguer (1 for TFT uart)

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 1  // -1 for USB 6 pour esp sur la carte mere
// Je branche les 3 cable EXP1 EXP2 et l'uart de cette facon je peux basculer entre le mode`

tbojim1975, 2020-03-05:
you are doing great in English :-) Thank you for your help

tbojim1975, 2020-03-06:
line 99 of the file
`/ **

Select the serial port on the board to use for communication with the host.
This allows the connection of wireless adapters (for instance) to non-default port pins.
Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
: [- 1, 0, 1, 2, 3, 4, 5, 6, 7]

/
#define SERIAL_PORT -1 <------ -1 worked
/ **

Select a secondary serial port on the board to use for communication with the host.
This allows the connection of wireless adapters (for instance) to non-default port pins.
Serial port -1 is the USB emulated serial port, if available.
: [- 1, 0, 1, 2, 3, 4, 5, 6, 7]

/
#define SERIAL_PORT_2 3 <------------ 3 worked

And my braudrate is set to 115200

Thank you for all your help it worked

UnderXirox, 2020-03-06:
Nice to have helped you

alieneila, 2020-03-19:
I have tried -1 and 3 as the Serial ports, as well as other combinations I've seen people suggest,
TFT and EXP 1 and EXP 2 connected - it works in Marlin mode, but does not work in touch screen mode - "No printer attached" - anyone have any suggestions?

tbojim1975, 2020-03-19:
Where the serial port -1 is. go a few more lines down from there and uncomment the other serial port_2 add a 3 so look like this "serial port_2 3" make sure b-rate is 115200

alieneila, 2020-03-19:
Sorry I wasn't clear... I already have PORT as -1 and PORT_2 as 3

tbojim1975, 2020-03-19:
I have an issue if anyone can help me that would be great. My equipment ender 3 bed/ Btt gtr 1.0/ btt tft 3.5 e3 v3.0 / 3in 1out NF hotend. Every time I try to use the manual bed leveling feature. The 4 points. It seems the points are set for a larger bed.  Because everytime I tried point 1 with my nozzle then 10 mm in on the bed then hit point 2 on the tft display. Then my nozzle courage try's to go further passed the bed like machine gun sound. Any ideas would be appreciated.

alieneila, 2020-03-20:
If your bed size in the firmware is set correctly, then you can search in configuration.h for LEVEL_BED_CORNERS and adjust the offsets.

nsmdigitalpc, 2020-03-20:
If you will Alieneila, can you please give me an example for an ender 3 
my bed size is 235 x 235
on the front left corner of my bed is Y: 8.7  X: 0.0 Z:0.0  
back right corner is X:115.0 Y:120.0 Z:0.0

defualt : 
#if ENABLED(LEVEL_BED_CORNERS)
  #define LEVEL_CORNERS_INSET_LFRB { 30, 30, 30, 30 } // (mm) Left, Front, Right, Back insets
  #define LEVEL_CORNERS_HEIGHT      0.0   // (mm) Z height of nozzle at leveling points
  #define LEVEL_CORNERS_Z_HOP       4.0   // (mm) Z height of nozzle between leveling points
  #define LEVEL_CENTER_TOO              // Move to the center after the last corner
#endif

nsmdigitalpc, 2020-03-20:
Is this what you mean?

// Add a menu item to move between bed corners for manual bed adjustment
#define LEVEL_BED_CORNERS

#if ENABLED(LEVEL_BED_CORNERS)
  #define LEVEL_CORNERS_INSET_LFRB { 0.0, 9.0, 114.0, 120.0 } // (mm) Left, Front, Right, Back insets 
 #define LEVEL_CORNERS_HEIGHT      0.0   // (mm) Z height of nozzle at leveling points
  #define LEVEL_CORNERS_Z_HOP       4.0   // (mm) Z height of nozzle between leveling points
  #define LEVEL_CENTER_TOO              // Move to the center after the last corner
#endif

alieneila, 2020-03-21:
Well, if your bed size isn't set correctly in Marlin then it'll go to the
wrong spot...

#define X_MAX_POS 200
#define X_MIN_POS -45
#define Y_MAX_POS 170
#define Y_MIN_POS -20
#define Z_MAX_POS 130
#define Z_MIN_POS 0

nsmdigitalpc, 2020-03-21:
my bed size is 235 x 235
on the front left corner of my bed is Y: 8.7 X: 0.0 Z:0.0
back right corner is X:115.0 Y:120.0 Z:0.0

Please help me understand this is. the Information i've sent before, is my off sets so can you give me an example

// The size of the print bed
#define X_BED_SIZE 235
#define Y_BED_SIZE 235

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 250

closed 2020-08-17 by bigtreetech