bigtreetech/BIGTREETECH-TouchScreenFirmware #359: [FR] IDEA: use x-macros

Andrei-Pozolotin, 2020-03-05:
@guruathwal

currently, there is much duplication and index management effort around:

* language labels and packs
[Vx.x.25/TFT/src/User/API/Language/Language.h](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/Vx.x.25/TFT/src/User/API/Language/Language.h)
[Vx.x.25/TFT/src/User/API/Language/Language.c](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/Vx.x.25/TFT/src/User/API/Language/Language.c)

* icon file names
[Vx.x.25/TFT/src/User/API/boot.h](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/Vx.x.25/TFT/src/User/API/boot.h)
[Vx.x.25/TFT/src/User/API/boot.c](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/Vx.x.25/TFT/src/User/API/boot.c)

there is a better approach based on on X_Macro:
https://en.wikipedia.org/wiki/X_Macro

compare to a working prototype:

* language labels and packs
[v-user-1/TFT/src/User/API/Language/Language.inc](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/TFT/src/User/API/Language/Language.inc)
[v-user-1/TFT/src/User/API/Language/Language.h](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/TFT/src/User/API/Language/Language.h)
[v-user-1/TFT/src/User/API/Language/Language.c](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/TFT/src/User/API/Language/Language.c)

* icon file names
[v-user-1/TFT/src/User/API/icon_list.inc](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/TFT/src/User/API/icon_list.inc)
[v-user-1/TFT/src/User/API/icon_list.h](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/TFT/src/User/API/icon_list.h)
[v-user-1/TFT/src/User/API/icon_list.c](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/TFT/src/User/API/icon_list.c)

basic idea is to define `structure.inc` once, and generate `pack`, `icon`, `label`, etc 
via each separate invocation of  `#include structure.inc`

guruathwal, 2020-03-10:
ok, I will try to include it in next PR.

Andrei-Pozolotin, 2020-03-10:
great.

currently there are 4 x-macro in the prototype:

* for user-defined `config.ini`
[v-user-2/TFT/src/User/API/config.inc](https://github.com/random-builder/bigtree-tft-code/blob/v-user-2/TFT/src/User/API/config.inc)

* for icon support
[v-user-2/TFT/src/User/API/UI/icon_list.inc](https://github.com/random-builder/bigtree-tft-code/blob/v-user-2/TFT/src/User/API/UI/icon_list.inc)

* for symbol support (a.k.a. `iconchar`)
[v-user-2/TFT/src/User/API/UI/symbol.inc](https://github.com/random-builder/bigtree-tft-code/blob/v-user-2/TFT/src/User/API/UI/symbol.inc)

* for language support
[v-user-2/TFT/src/User/API/Language/Language.inc](https://github.com/random-builder/bigtree-tft-code/blob/v-user-2/TFT/src/User/API/Language/Language.inc)

you can check them out in action:

* firmware snapshot
[v-user-2/DISK](https://github.com/random-builder/bigtree-tft-code/tree/v-user-2/DISK)

closed 2020-08-17 by bigtreetech