bigtreetech/BIGTREETECH-TouchScreenFirmware #360: [FR] IDEA: user managed config.ini

Andrei-Pozolotin, 2020-03-05:
@guruathwal

consider implementing user-managed configuration system, aka  "config.ini":

here is working prototype:

* user file format:
[v-user-1/DISK/config.ini](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/DISK/config.ini)

* developer config definition:
[v-user-1/TFT/src/User/API/config.inc](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/TFT/src/User/API/config.inc)

* current implementation detail:
[v-user-1/TFT/src/User/API/config.h](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/TFT/src/User/API/config.h)
[v-user-1/TFT/src/User/API/config.c](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/TFT/src/User/API/config.c)

Andrei-Pozolotin, 2020-03-06:
Example: `config.ini` in action:

#### default screen
[image]

#### config.ini section
```
#
# use case light M355 instead of rgb M150
#
[light_control]

#
level_1_icon  = RGB_White
level_1_label = Set 25%
level_1_gcode = M355 P064 ; 255 * 0.25
#
level_2_icon  = RGB_White
level_2_label = Set 50%
level_2_gcode = M355 P128 ; 255 * 0.50
#
level_3_icon  = RGB_White
level_3_label = Set 75%
level_3_gcode = M355 P192 ; 255 * 0.75
#
level_4_icon  = RGB_White
level_4_label = Set 100%
level_4_gcode = M355 P255 ; 255 * 1.00
#
level_7_icon  = RGB_Off
level_7_label = Reset
level_7_gcode = M355 P000
```

#### custom screen result
[image]

firmware: https://github.com/random-builder/bigtree-tft-code/tree/custom-marlin/DISK

thisiskeithb, 2020-03-28:
I haven't tested this, but do settings persist on reboot/without an SD card inserted after the initial config is loaded?

If so, it'd be easier for non Ender-3 users (the current default) to change core settings.

Andrei-Pozolotin, 2020-03-28:
yes, stored in spi-flash, please read the code :-)

thisiskeithb, 2020-03-28:
> yes, stored in spi-flash, please read the code :-)

I just wanted to confirm that it works as coded in your testing/you tested without an SD card.

Andrei-Pozolotin, 2020-03-28:
yes

closed 2020-08-17 by bigtreetech