bigtreetech/BIGTREETECH-TouchScreenFirmware #361: [FR] IDEA: remote firmware deploy with build.py

Andrei-Pozolotin, 2020-03-05:
@guruathwal 

consider inclusion of `build.py` script to automate firmware build and image deployment:

* tool folder:
[v-user-1/buildroot/user](https://github.com/random-builder/bigtree-tft-code/tree/v-user-1/buildroot/user)

* build entry point:
[v-user-1/buildroot/user/build_USER.py](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/buildroot/user/build_USER.py)

* platfromio integration:
[v-user-1/buildroot/user/setup.py](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/buildroot/user/setup.py)

* remote host deploy script:
[v-user-1/buildroot/user/setup_disk.py](https://github.com/random-builder/bigtree-tft-code/blob/v-user-1/buildroot/user/setup_disk.py)

oldman4U, 2020-10-20:
Could you please be so kind and close this ticket. Your feature request is covered in the pinned Feature Request ticket #1177 

Thank you

closed 2020-11-02 by Andrei-Pozolotin

oldman4U, 2021-01-01:
Hi. Hope you are doing fine.

Could you please check if this solves your request: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1463

Thank you and happy new year