bigtreetech/BIGTREETECH-TouchScreenFirmware #362: printing menu and echo

JediRAMA, 2020-03-05:
Hello. 
Any chance to change printing menu? Im using command every lyer M117 to see the end timer of printing job. And i see alltime this at my screen
[image]
[image]

Is it possible to add echo in printing menu like this example? And change a bit menu 

[image]

oldman4U, 2020-09-11:
Hello Jedi.

Please close this ticket in case the issue is solved in the meantime.

May the force be with you

closed 2020-10-13 by JediRAMA

oldman4U, 2021-01-12:
Hi.

Could you please check #1410 which could solve the reported feature request.

Thank you

digant73, 2021-07-27:
@oldman4u also this bug was fixed. M117 are now reported in status area and notification area as both toast and notification. Time Left is also properly handled by the PR implemented M73. So this bug can also be removed from the list