bigtreetech/BIGTREETECH-TouchScreenFirmware #365: [FR] IDEA: move language.c into language.ini

Andrei-Pozolotin, 2020-03-06:
@guruathwal

please consider moving language support form code into configuration,
in other words, from `mcu flash` into `language.ini` and then into `spi flash`

similarly to how it is done for use-managed configuration support with `config.ini`
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/360

major benefits:
* for users: can fix and customize their language themselves
* for developers: no need to maintain languages they do not understand :-)

guruathwal, 2020-03-10:
I will recommend keeping the Language part in the firmware because each language has a different set of alphabets and the length of words changes with each language, there are words as short as one character and sentences longer than 50 characters.
The data in the `SPI Flash` can only be read by knowing the start address of the bytes and the length of the bytes to read. Due to this, the length of all the keywords will be the same and limited to the predefined length in the firmware.
This will result in overwriting if the keywords are changed to words with longer lengths like `Move` to `Movement`.

Andrei-Pozolotin, 2020-03-10:
I think this issue can be resolved differently. 

Please take a look on current `config.ini` implementation

* full text of `config.ini` is stored in `spi flash` 
[v-user-2/TFT/src/User/API/boot.h](https://github.com/random-builder/bigtree-tft-code/blob/v-user-2/TFT/src/User/API/boot.h#L36)

*  `config.ini` is parsed on demand and placed in `mcu heap`
[v-user-2/TFT/src/User/API/config.h](https://github.com/random-builder/bigtree-tft-code/blob/v-user-2/TFT/src/User/API/config.h#L15)

One benefit of `language.ini` is that there is no need to load 
all 14 (currently) or 140 (in the future) languages in `spi flash`, 
only 1 or 2 languages people actually use at a given location

guruathwal, 2020-03-10:
Did you test on only the TFT35 V3?

Only the V3 hardware version has the powerful **STM32F207VCT6** MCU with 132kB RAM. The **TFT24 V1.1** is based on **[STM32F105RC](https://www.digikey.com/product-detail/en/stmicroelectronics/STM32F105RCT6/497-8918-ND/2074112)** which have 64kB RAM while **TFT35v2** is based on **[STM32F103VC](https://www.digikey.com/product-detail/en/stmicroelectronics/STM32F103VCT6TR/497-17393-2-ND/2214607)** which has only 48 kB RAM.

You need to check the heap size on all hardware variants while printing from the SD card and USB flash drive.
Also, try loading text with the length equivalent to the length of the total keywords and then check its effect on all the devices thoroughly.

I can agree with you on this only if it does not affect the devices with low RAM and after that also if we are considering just two languages.
First one English, which should be hard-coded in the firmware and second which can be one of the supported languages to be loaded in SPI Flash.
We need to be fully sure that this won't affect the devices with lower RAM.

Andrei-Pozolotin, 2020-03-10:
additionally:

since `spi flash` 8 MB limit is also coming up (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/364) and unicode font consumes 50% of flash
```
TFT35/font/word_unicode.fon
4.5 MiB (4,718,592)
```

then next obvious idea is to partition `word_unicode.fon` 
by `script` or by `block` or by `plane` or by "some other group criteria"
https://www.w3.org/International/talks/0505-unicode-intro/
https://en.wikipedia.org/wiki/Unicode_block
```
unicode_group_1.fon // alphabetic, europe
unicode_group_2.fon // syllabic, cjk
...
```
and load  a pair of `unicode_FONT.fon` and `language_DICT.ini` only on demand, 
for example:

```
// enable English:
language_font = unicode_group_1.fon
language_dict = language_EN.ini
```
```
// enable Spanish:
language_font = unicode_group_1.fon
language_dict = language_ES.ini
```
```
// enable Japanese:
language_font = unicode_group_2.fon
language_dict = language_JP.ini
```

that approach could easily free up some 75% of flash (when using 4 partitions)
and then to allow to store extra 350 (95x95x2) icons :-)

oldman4U, 2020-10-09:
@guruathwal Will you close this ticket automatically with your latest PR?

PR #1138 (2020-10-19): Update icons, Fix ABL G-code sending on mode change, Fix text overlap in update screen, Move language translations to SPI flash to reduce main memory usage, Fix other Issues/Bugs
- Update icons.
- Fix ABL G-code sending on Mode change.
- Fix text overlap in update screen.
- Fix value step button not displaying the correct value in heat menu.
- Move language translations to SPI flash to reduce main memory usage and allow switching/updating language translations without recompiling the firmware. (#365)
  - To update/ add a second language, the user needs to copy the required `language_xx.ini` fi [... cut]
closed 2020-10-19 by bigtreetech