bigtreetech/BIGTREETECH-TouchScreenFirmware #366: [FR] Some sort of auto sleep for touch mode

ReubenBTalbott, 2020-03-07:
I don't know if this is possible to do, but could the screen when in touch mode after a set amount of  time turn off the backlight until touched? 
I use a relay to shutdown my printer's main board after printing, but keep the screen powered with a  buck converter, and when I want to print again I can power the mainboard back on with the screen, so when the printer is off the screen is still on and doing nothing.

dziabi3, 2020-03-07:
Apparently this feature is to be added in the next update. I read it once here.

bkzspam, 2020-03-08:
I agree, this can be great feature. There's a should be an option called as "Screen auto off time"

bigtreetech, 2020-04-10:
Hello, it's done in #474, you can download the lastest version firmware to test

ReubenBTalbott, 2020-04-11:
Awesome!

closed 2020-04-11 by ReubenBTalbott