bigtreetech/BIGTREETECH-TouchScreenFirmware #368: [BUG] Printing still resumes in screen even printer halted.

bkzspam, 2020-03-08:
If printer halt when printing, screen continues to show as printing is resuming. (Example situation: When MINTEMP protection triggered and printer halted.)

Expected behaviour: Screen should go back to main menu and say "Printer halted, reset the printer." or asks "Reset printer?" (It connected with reset pin to board so this is possible.)

mikefromups, 2020-03-11:
Not sure if this is along the same lines, but have recently had an issue with a file that printed fine the day before.  Then, when printing the file the following day, the printer would halt just as soon as it started.  Maybe 1 second and the progress went to 40 % and all stepper movement stopped as well as the heaters remained on.  The printer had to be reset before control resumed and then the heater could be stopped.  After several attempts of the same file with similar results I decided to try other files and most printed fine.  I thought maybe the file was corrupt so after slicing again, the print worked fine.  My concern is the fact that with a bad gcode file, the printing could halt and the heaters stay on indefinitely.  I hope this would only be a frustrating experience and not a fire hazard since thermal runaway was not the issue.

oldman4U, 2020-10-09:
Have you tried the latest builds?

closed 2020-11-27 by bigtreetech