bigtreetech/BIGTREETECH-TouchScreenFirmware #371: Hungarian language

AntoszHUN, 2020-03-09:
**The problem**
Hi! I use the SKR TFT35 V3.0 display. I have no problem with it. The only problem is that there is no Hungarian language on it. As it contains 11 languages, Hungarian would fit in.

**Solution**
I made the Hungarian language. they could install it in the next update, which would be very good for many Hungarian users.

**Alternative**
Here is the menu system in Hungarian. https://www.youtube.com/watch?v=_sTpq-FcNio

[image]

Acenotass, 2020-03-17:
Hi!
To implement the Hungarian language you need to make a Pull Requests.

AntoszHUN, 2020-03-17:
All right. :)

Acenotass <notifications@github.com> ezt írta (időpont: 2020. márc. 17., K
10:35):

AntoszHUN, 2020-03-17:
I can't create a Pull Request. :(

AntoszHUN, 2020-03-17:
[image]

What is it? I can't create a Pull Request. :(

Acenotass, 2020-03-18:
Please attach all your translate files.
Or do it himself.
https://www.youtube.com/watch?v=8lGpZkjnkt4

AntoszHUN, 2020-03-18:
Hi!

I uploaded it to Drive. Here is the sharing link. I couldn't compress the folder. This is all firmware because there is not just a language_hu.h file but I have translated everything. There is a (Read_el) file in the folder. I wrote everything down there. If you have any questions, do not hesitate to contact me.

Thanks for the videos. Unfortunately, I don't really understand what they're doing in it. But once we can solve it ...

In any case, I've attached the entire firmware sharing link.

Sincerely!
AntoszHUN

https://drive.google.com/drive/folders/14sPiJP6DvAy8s0EAWiZDwkouchvXqPnu?usp=sharing

bigtreetech, 2020-03-20:
merged in #424 ， thank you!

AntoszHUN, 2020-03-20:
Thank you very much!

AntoszHUN, 2020-03-20:
Is only the language_hu.h file inserted? Or will the others be included? Maybe I can find out which files I have translated?

AntoszHUN, 2020-03-20:
Hi! English cannot be selected in the display menu. Integration successful?
[image]

Acenotass, 2020-03-20:
Yes, everything works well, the Hungarian language has appeared.
Download the latest firmware, fonts, and Icons.

AntoszHUN, 2020-03-21:
All right. Works with minor modifications left out of configuration.h. Please make the following correction!
I supplemented it with HUNGARY.

/ **
  * Default LANGUAGE
  *
  * Select the language to display on the LCD. These languages are available:
  * ENGLISH, CHINESE, RUSSIAN, JAPANESE, ARMENIAN, GERMAN, HUNGARY, CZECH, SPAIN, FRENCH, PORTUGUESE, ITALIAN, POLISH, SLOVAK, DUTCH
  * /

Thank! Sincerely!
Antosz

closed 2020-03-21 by AntoszHUN