bigtreetech/BIGTREETECH-TouchScreenFirmware #372: [BUG] Can't use "classic" menu style

thisiskeithb, 2020-03-10:
### Description

Commenting out `UNIFIED_MENU` (to use "classic" mode) & icons from the non-unified menu folder results in a broken menu as if `UNIFIED_MENU` is still being used.

### Steps to reproduce

1. Use the [config](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/b7d736eeab490563358726870fb945bc8ce49939/TFT/src/User/Configuration.h) from b7d736e, and comment out `UNIFIED_MENU`.

**Expected behavior**
Working Classic UI

**Actual behavior**
Broken classic UI main screen:
[image]

All the sub-menus look fine (here's just one example):
[image]

### Hardware Variant
TFT35 3.0 and TFT35 E3 3.0

### TFT Firmware Version & Main Board Firmware details
b7d736e, TFT_35_V3.0.25.2 Mar 9 2020

guruathwal, 2020-03-10:
It is the Status Screen. BTT has added the Status Screen to the Classis Menu also. You will have to use the status screen icons from the Unified menu folder.

thisiskeithb, 2020-03-10:
@bigtreetech Was this intentional? If the classic menu is going away, more cleanup is required.

thisiskeithb, 2020-03-13:
Closing since PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/390 reverted the status screen for "Classic UI".

closed 2020-03-13 by thisiskeithb