bigtreetech/BIGTREETECH-TouchScreenFirmware #373: [BUG] Disabling ALTERNATIVE_MOVE_MENU breaks compilation

thisiskeithb, 2020-03-10:
### Description

Disabling `ALTERNATIVE_MOVE_MENU` breaking compilation.

### Steps to reproduce

1. Use the [config](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/b7d736eeab490563358726870fb945bc8ce49939/TFT/src/User/Configuration.h) from b7d736e, and comment out `ALTERNATIVE_MOVE_MENU`.
2. Compile
3. Observe errors:

```prolog
...

Compiling .pio/build/BIGTREE_TFT35_V3_0/src/src/User/Menu/Move.o
Compiling .pio/build/BIGTREE_TFT35_V3_0/src/src/User/Menu/Parametersetting.o
Compiling .pio/build/BIGTREE_TFT35_V3_0/src/src/User/Menu/Popup.o
TFT/src/User/Menu/Mode.c: In function 'menuMode':
TFT/src/User/Menu/Mode.c:106:11: warning: unused variable 'nowEncoder' [-Wunused-variable]
   int16_t nowEncoder = encoderPosition = 0;
           ^~~~~~~~~~
TFT/src/User/Menu/Move.c: In function 'menuMove':
TFT/src/User/Menu/Move.c:251:32: error: 'XGCODE_UP' undeclared (first use in this function); did you mean 'XGCODE_DEC'?
                       storeCmd(XGCODE_UP, item_move_len[item_move_len_i]);  
                                ^~~~~~~~~
                                XGCODE_DEC
TFT/src/User/Menu/Move.c:251:32: note: each undeclared identifier is reported only once for each function it appears in
TFT/src/User/Menu/Move.c:254:32: error: 'YGCODE_UP' undeclared (first use in this function); did you mean 'XGCODE_UP'?
                       storeCmd(YGCODE_UP, item_move_len[item_move_len_i]);
                                ^~~~~~~~~
                                XGCODE_UP
TFT/src/User/Menu/Move.c:257:32: error: 'ZGCODE_UP' undeclared (first use in this function); did you mean 'YGCODE_UP'?
Compiling .pio/build/BIGTREE_TFT35_V3_0/src/src/User/Menu/PowerFailed.oi]);

Compiling .pio/build/BIGTREE_TFT35_V3_0/src/src/User/Menu/PreheatMenu.o
                                ^~~~~~~Compiling .pio/build/BIGTREE_TFT35_V3_0/src/src/User/Menu/Print.o
~~
                                YGCODE_UP
TFT/src/User/Menu/Move.c:268:32: error: 'XGCODE_DOWN' undeclared (first use in this function); did you mean 'XGCODE_DEC'?
                       storeCmd(XGCODE_DOWN, item_move_len[item_move_len_i]);  
                                ^~~~~~~~~~~
                                XGCODE_DEC
TFT/src/User/Menu/Move.c:271:32: error: 'YGCODE_DOWN' undeclared (first use in this function); did you mean 'XGCODE_DOWN'?
                       storeCmd(YGCODE_DOWN, item_move_len[item_move_len_i]);
                                ^~~~~~~~~~~
                                XGCODE_DOWN
compilation terminated due to -fmax-errors=5.
Compiling .pio/build/BIGTREE_TFT35_V3_0/src/src/User/Menu/Printing.o

...

```

**Expected behavior**
No compile errors

**Actual behavior**
Compile errors

### Hardware Variant
TFT35 3.0 and TFT35 E3 3.0

### TFT Firmware Version & Main Board Firmware details
b7d736e, TFT_35_V3.0.25.2 Mar 9 2020

### Additional Information

I can change the variables to `*_INC`/`*_DEC` to allow a successful compile, but then the move buttons no longer work.

guruathwal, 2020-03-10:
I can confirm this. It got messed up after with PR #311 by @bcmob

guruathwal, 2020-03-10:
@thisiskeithb The current layout of the Move menu is not made for use encoders. I suggest the encoder support should be removed from the Move menu until a new layout is added to be compatible with encoders.

thisiskeithb, 2020-03-10:
It also seemed related to #338, but the commit history with some of these PRs are quite a mess and difficult to follow when changes are re-merged.

I’ve reverted back to 5777f41 for now since that still works.

PR #389 (2020-03-13): clean up move.c & default config disable AUTO_SAVE_LOAD_LEVELING_VALUE
### Requirements

* Filling out this template is required. Pull Requests without a clear description may be closed at the maintainers' discretion.

### Description

### Benefits

### Related Issues
fixed #373
closed 2020-03-13 by bigtreetech