bigtreetech/BIGTREETECH-TouchScreenFirmware #374: [FR] More "Classic UI" testing

thisiskeithb, 2020-03-10:
I still prefer the "Classic UI", but several recent changes have made it difficult to use (and would be a non-starter for most users).

Please do more testing with the "Classic UI" enabled with PRs (and flash on real hardware) to make sure it still works correctly.

Hukuma1, 2020-03-10:
I would argue the complete opposite that the unified menu is a lot better. What makes you like the Classic UI better? Is there a certain feature missing from it that unified does not have? Or are you talking strictly just about the icons?

The fact we still have 2 menu layouts leads to end user confusion and code debugging fragmentation. I often say choice is great, but in this case I see it holding the firmware back more than helping.

thisiskeithb, 2020-03-10:
> What makes you like the Classic UI better?

For the time being, the classic UI just works with no fuss.

thisiskeithb, 2020-03-10:
This project now supports MKS TFTs, so should that be dropped too?

Hukuma1, 2020-03-10:
> For the time being, the classic UI just works with no fuss.
I mean, that's how I currently have unified working? That's why I ask what is possibly missing from unified that you think could be improved for real use case. I can only comment on the unified UI's helpfulness for someone who wants to click print and have it do what is on the screen with the least amount of interaction. The ability to see all the most pertinent information at a quick glance on one screen is the biggest advantage for me personally.

> This project now supports MKS TFTs, so should that be dropped too?

I don't know what this means. I've been out of the loop. Someone ported BTT TFT firmware to support another device?

thisiskeithb, 2020-03-10:
> The ability to see all the most pertinent information at a quick glance on one screen is the biggest advantage for me personally.

Thats great for you, but I prefer the Classic UI and until @bigtreetech confirms that it’s being removed, it should still be supported and work as did before instead of this broken mashup.

> I don't know what this means. I've been out of the loop. Someone ported BTT TFT firmware to support another device?

Yes. See https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/329

Hukuma1, 2020-03-10:
Okay, so just to be clear, I sincerely asked why you prefer it, and your answer is "just because." Well yeah, I hope @bigtreetech then gives an update on their plan with this. Since you code stuff, maybe you could help with keeping support of it?

Cool I guess regarding the MKS device. Although I bought the BTT display for the hardware, not for its software. So yeah, not sure what dropping MKS support would have to do with my BTT device purchase. In the same way running Windows on a Hackintosh, I would hope people aren't buying a Macbook with the sole purpose of running Windows 10. Does MKS not have its own firmware development team?

thisiskeithb, 2020-03-10:
I was trying to point out that simplifying code by dropping features, yet adding support for non-BTT boards doesn’t make sense if “code debugging fragmentation” is your concern. Either maintain backward compatibility or don’t.

> Okay, so just to be clear, I sincerely asked why you prefer it, and your answer is "just because."

I said: “For the time being, the classic UI just works with no fuss.“

To expand on that, a lot of development has gone into the Unified theme (hats off to @guruathwal and others!) and I like where it’s headed, but that also means quirks and bugs. So until that code and features mature, I prefer the original UI.

> Since you code stuff, maybe you could help with keeping support of it?

I’m waiting on confirmation from BTT of what the plans are before I put in effort into fixing anything.

> Does MKS not have its own firmware development team?

MKS’ firmware is not open source.

At any rate, I’m just asking for more testing of existing features since they are still available in the current codebase.

Hukuma1, 2020-03-10:
So if the bugs and quirks are resolved from Unified, you'll prefer it over classic? Sounds like you should just invest the time and effort into helping the wonderful @guruathwal with unified.

I can't help but think that Classic = Windows 95 and Unified = Windows 10. I don't see how supporting both is helpful for development. That's the fragmentation I am talking about. I am obviously biased towards the newer as stated above. But was genuinely curious why classic was better in your opinion. If it's the bugs you're talking about, let's just squash the bugs! :)

As for the MKS thing. Sorry to hear, but super cool that someone made it work! But again, diverting resources to an outside company's product... would be weird? If anything, BTT would focus on their own product to show how superior it is. They ultimately probably (read: opinion) want to sell their screens, not just get downloads of their free firmware.

But yes, fully agreed that more testing of the new code is definitely needed. I do small prints myself, so I can only test for my specific use.

thisiskeithb, 2020-03-10:
With a simple graphic update, Classic UI looks great:
[image]

Hukuma1, 2020-03-10:
Absolutely agreed.

guruathwal, 2020-03-11:
The first reason for the bugs is when the developers push PR without fully checking the firmware with all the Features set to both Enabled and Disabled. Most of the PRs are for small tweaks for one specific hardware variant which may result in unintentional bugs for other variants that are left unattended.

The second reason being the two variants of menu styles to be maintained, due to which maintaining the code requires double effort.

I originally intended to keep the classic menu for some time until the Status screen was bug-free, similar to the List mode in the Feature Setting menu. But BTT added the Status screen to the classic menu which led to this confusion and problem.

I am in favor of phasing out the classic menu and keeping only one style. But it is the whole community that has to decide which one they like to keep. Why not create a poll and see what everyone is in favor of?

bigtreetech, 2020-03-12:
> With a simple graphic update, Classic UI looks great:
> [image]

I added the "status screen" in the classic UI because I think the information of the "status screen" is important,  but I ignores the complexity brought by "status screen", so do you think that I need to delete "status screen" in the classic UI and keep the classic UI as simple as possible?

thisiskeithb, 2020-03-12:
> so do you think that I need to delete "status screen" in the classic UI and keep the classic UI as simple as possible?

I’d recommend separating the two menu styles as they were before (and fix the current state of the Classic UI. See  https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/372) or get rid of the Classic UI completely if all development moving forward be on the Unified menu.

guruathwal, 2020-03-12:
I will recommend having only one style because with two types it will be more difficult to maintain and add features to both of them.

thisiskeithb, 2020-03-12:
> I will recommend having only one style because with two types it will be more difficult to maintain and add features to both of them.

A poll could be created in the main BigTreeTech group to see what the users prefer. And of course, the project could always be forked from an earlier commit if they prefer the old style.

thisiskeithb, 2020-03-12:
A possible third option: Port the Classic UI bitmaps over (and add any missing ones) as an icon pack to the Unified UI and remove the Classic UI.

This might be the best “compromise”, by allowing more code cleanup/removal, easier development, and as a bonus: simplify the readme and update process for users.

thisiskeithb, 2020-03-13:
Closing since my issues have been addressed, but more testing with both modes would still be appreciated when new features are rolled out.

closed 2020-03-13 by thisiskeithb