bigtreetech/BIGTREETECH-TouchScreenFirmware #376: Using encoder with touchscreen on TFT24

edcasati, 2020-03-10:
The default touchscreen GUI does not seem to use the rotary encoder at all.
Sometimes when moving an axis on a 3D printer or CNC, it takes multiple presses of the screen to complete the movement, and that can be difficult to do reliably. It would be nice to be able to select an axis, select a motion quantity, and then just turn the encoder the necessary number of clicks to move the desired amount.
For example, if I want to move 'X" 80mm, I have to set the distance to 10mm and press the appropriate screen button 8 times. It would be much easier to select the amount of motion (10mm), the axis to move (X, Y, X+Y, Z) and then just turn the encoder 8 clicks in the appropriate direction.
Is there example code where the rotary encoder is read while in graphical mode?

bigtreetech, 2020-03-14:
hello, this feature has be merged in #311 , you can test it now

angry-elf, 2020-03-24:
Is it already in binary build? I'm using TFT24_V1.1.25.2 (jan 31 2020) from this repo and seems encoder still not working.

oldman4U, 2020-09-11:
@bigtreetech 

Please close this ticket

Thank you

closed 2020-12-15 by stale[bot]