bigtreetech/BIGTREETECH-TouchScreenFirmware #377: [BUG] Since new printing screen was implemented, icons are not correctly shown

javicarrera, 2020-03-10:
### Description

Since I updated yesterday my TFT24 v1.1 and TFT35 V3.0, the printing screen is a mess, the icons are not properly shown, looks like corrupted data.

### Steps to reproduce

1. Update to newest firmware.
2. Print, get to the printing screen.

**Actual behavior**
The icons look corrupted.
It seems all icons starting with the name "printing" and "global" are corrupted.

### Hardware Variant
TFT 24 v1.1 and TFT35 v3.0
### TFT Firmware Version & Main Board Firmware details
25.2 (from yesterday 09/03/2020

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

mikefromups, 2020-03-11:
Not knowing your level of expertise, so just a suggestion/comment.  Did you also copy the fonts and icons, included with the update, to your flash drive?  Some updates require new icons or fonts to be reloaded.

JWolfJr, 2020-03-11:
Thank you for bringing this to my attention @mikefromups. I've been irritated why my icons were messed up and or missing. Just had to reload my folder with the bmp images. So simple, I feel extra dumb! Thank you for posting this.

javicarrera, 2020-03-11:
Hello thank you all. Reloading the folder with the images didn't solve the problem. I had to compile myself the firmware to make it work, now it works! So I assume it is something wrong with the bin file.

closed 2020-03-11 by javicarrera