bigtreetech/BIGTREETECH-TouchScreenFirmware #378: Should the printing screen be causing slight pausing of print activity?

mikefromups, 2020-03-11:
Recently during a small print job I decided to increase the Speed by changing from 100% to 150 or 200%.  While on the "printing" screen I noticed a slow down of the printing compared to other screens.  I am assuming this is because of the constant updates of the live feedback to the screen.  I had no idea it was that noticeable until you go to 200%, and with a small footprint (30 mm x 30 mm) and lots of direction changes you could really hear the difference.   

Is this due to the processor capabilities or the way the tft is connected and it's baud rate?  Just wondering if the newest version tvt35 v3 will be better

bigtreetech, 2020-03-13:
yeah. It may take too much time to refresh the data, because the ICON of the new version needs to be redrawn, If it is for this reason, we are planned to cancel the redraw icon later, or process the receiving and sending gcode commands in the timer interrupt, and process the UI display in the frontend loop

guruathwal, 2020-03-14:
@mikefromups can you share your current TFT firmware version and Date. and also your Marlin Firmware Version

closed 2020-08-19 by bigtreetech